TheSpaghettiDetective/obico-server #12: Configure an smtp container in docker-compose with reasonable default so that users don't have to have a smtp service to enable email

kennethjiang, 2019-02-26:
Example: https://github.com/getsentry/onpremise

pattonwebz, 2019-03-25:
Email deliverability is somewhat complicated, I would really recommend people use an existing SMTP server for enabling emails if they want to try make sure they receive them when they need to.

Something like https://alternativeto.net/software/mailhog/ could work here (and can forward to external SMTP service is available). I have seen docker images for this and I am fairly certain we could have it configured to read `env` variables at `up`time so that we could pass in config details without need to manually configure it after it's first started.

arcreigh, 2019-11-19:
I am strongly against this for a couple of reasons.

1. Port 25 is reasonably blocked by Residential ISP's and many (comcast) do not give any documentation on how to use their SMTP servers for forwarding.
2. E-Mail is difficult to keep secure and would require a lot of effort on the dev's part to keep up with the ever changing e-mail threat environment, although if we are looking at a send only domain it would not be too terrible to configure.
3. Most sensible mailbox hosts will not accept e-mail from residential IP's as they are utilizing RBL's (Realtime Black Lists) 
4. Those who do not know how to properly maintain an SMTP server really should not have one.

marktoml, 2019-11-20:
Consider forcing SSL and OAUTH.

kennethjiang, 2019-11-20:
@arcreigh @marktoml Thank you for chiming in! I agree that most users shouldn't run their own smtp server. Since it looks like most of them have been able to configure email using their own gmail, I'm closing this issue in favor of using gmail as delivery.

closed 2019-11-20 by kennethjiang