TheSpaghettiDetective/obico-server #13: Should not send images when printer is inactive

Etumos, 2019-02-27:
The plugin connects to 'app.thespaghettidetective.com' once every 10 seconds to send only one picture from the webcam for the server to run algorithm on. In a 12 hour period the octopi connected to 'app.thespaghettidetective.com' approximately 11K times, while the printer was idle. 

The plug in should only function when octopi senses a print.

kennethjiang, 2019-02-28:
@Etumos Thank you for submitting this issue.

We talked about it and thought the plugin should still send images while printer is not printing, at a much reduced rate (probably 1 image every 1 or 2 minutes).

The reason is, if not image is sent, the "My Printer" page will show no hint that the plugin is connected and working as expected. Showing an image that is reasonably recent (up to 2 minute behind) will serve as a visual hint that the plugin is working but the printer is idle.

Let me know what you think.

marktoml, 2019-02-28:
I like that solution. Sometimes I want video even if I am not printing.

Etumos, 2019-02-28:
I appreciate that you guys hoped on this so quickly. however I think the same thing can be accomplished without sending an image. Is there some reason you could not just ping the octoprint server periodically and show a red / green light to represent connectivity. I get what the product is geared for, but taking pictures when not needed seems to be a needless privacy breach.

kennethjiang, 2019-03-01:
@Etumos In another OctoPrint plugin I built (https://plugins.octoprint.org/plugins/anywhere/) there is such status indicator. However, from the feedback of hundreds of users it was obvious to me that human respond much more intuitively to a picture than an indicator or status text. For example, a red dot or "disconnected" text can mean either "pi lost internet connection" or "printer is not connected to pi". I actually had to add hint text to differentiate these 2 situations but I still keep on getting emails about them. Whereas no one ever asked about the picture as long as it's showing a reasonably recent one.

Since the plugin is sending pictures when printer is printing, I don't think doing the same when printer is inactive amount to a different level of privacy problem. In particular this project is open source from front to back. Nothing to hide.

I just release version 0.3.0 that includes the reduced rate to send images. I'll close this issue for now. If you still think it's a problem feel free to re-open it.

closed 2019-03-01 by kennethjiang