TheSpaghettiDetective/obico-server #15: Building/rebuilding the so file

sqrtroot, 2019-03-02:
I was curious if you could share some details about the actual nn running. How was it generated and is it possible to rebuild the nn ourself?

IvanRybnikov, 2019-03-02:
That so file is the only platform-dependent part of this project. Therefore it's a bit tricky to have a generic way to build it for all platforms...

At the high level, `model.so` is just the `darknet.so` file built from [this repo](https://github.com/AlexeyAB/darknet). 

It'll be great if someone can help us figure out a universal way to build it for different platforms (Linux, Windows, Mac) without and without GPU support, and document the process.

foxabilo, 2019-03-18:
Willing and able windows 10 based victim here, have compiler, will debug for food.

kennethjiang, 2019-03-18:
Lol! Please feel free to take a stab at it. Please take a look at #19 in which @andylsun manually re-built it with openmp support.

Ideally we want to automate this process as much as we can, probably by running a command like `docker run thespaghettidetective/ml_api rebuild-so-file.sh`.

Your help will be much appreciated!

kennethjiang, 2019-12-19:
There have been multiple users who managed to rebuild `model.so` on different platforms. Anyone who runs into issues can come to our discord chat room and ask for help: https://discord.gg/NcZkQfj

closed 2019-12-19 by kennethjiang