TheSpaghettiDetective/obico-server #19: model.so is not compiled with multiple thread/OPENMP support

andylsun, 2019-03-06:
The current version of model.so that is provided in The Spaghetti Detective only runs on a single CPU core when not using GPU's. This is not very efficient.

I cloned https://github.com/AlexeyAB/darknet

modified Makefile Lines 6 and 7

OPENMP=1
LIBSO=1

Built libdarknet.so

Copied it to TheSpaghettiDetective/ml_api/bin and replaced the existing model.so with the new file.

Now TSD runs multi threaded and is much quicker on a system with lots of CPU cores but not Nvidia GPU.

[image]

kennethjiang, 2019-03-06:
@andylsun just tested it and it was 3x faster on my Mac. Awesome job. Just pushed the new .so with openmp enabled.

closed 2019-03-06 by kennethjiang

andylsun, 2019-03-07:
darknet seems to scale pretty nicely. Using 20 cores in a burst :)

top - 08:52:04 up 4 days, 23:01,  1 user,  load average: 4.03, 3.00, 1.95

Tasks:* 1073 *total,*   1 *running,* 1072 *sleeping,*   0 *stopped,*   0 *
zombie

%Cpu(s):* 22.8 *us,*  0.1 *sy,*  0.0 *ni,* 77.1 *id,*  0.0 *wa,*  0.0 *hi,*
0.0 *si,*  0.0 *st

KiB Mem :* 26373049+*total,* 21675160+*free,* 27545952 *used,* 19432932 *
buff/cache

KiB Swap:*  4194300 *total,*  4194300 *free,*        0 *used.* 25585892+*avail
Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND

52567 root      20   0   10.4g 732460  29872 S  1993  0.3 589:13.42 python

52668 root      20   0  286980 119932  14660 S  13.7  0.0   1:39.28
gunicorn

21594 root      20   0 3382376  72668  23936 S   2.0  0.0 247:48.89
containerd

*21847 root      20   0  162980   5492   3764 R   2.0  0.0   0:00.17 top
                         *

52188 root      20   0  240896  72976  16048 S   1.0  0.0   2:26.60 celery

    1 root      20   0  198040  12228   3928 S   0.0  0.0   0:51.28 systemd

    2 root      20   0       0      0      0 S   0.0  0.0   0:00.17
kthreadd

kennethjiang, 2019-03-07:
1993% CPU. Nice!
 
Have you changed the code to run `ml_api` in multiple threads? Or it was just 1 thread (the default of `ml_api`) that is using all 20 cores ?