TheSpaghettiDetective/obico-server #22: New printer flow is still too complicated for new user. Simplify it!

kennethjiang, 2019-03-13:


lukeslaboratory, 2019-04-14:
I propose that once you sign up on the website and create your account, you can point your individual octoprint setup to whatever server you choose and enter your information there - no secret key, no other nonsense - Give it a name, a registered email, and the destination, and let the browser-side handle the confirmation handshake (A window that goes HEY! Is this your printer? confirm your plan for this printer)

kennethjiang, 2019-04-14:
Can you elaborate your design a bit more @lukeslaboratory ? Currently a secret token is kept in both OctoPrint plugin configuration and the server. Whenever the plugin needs to send a pic or print status to the server, it passes the secret token in HTTP header. The server would then match that the token in HTTP header to the one in the database to make sure the request truly comes from the right plugin as it claims.

If your design doesn't include a secret token (lmk if I mis-interpret your design), how can the server make sure an incoming request should be trusted?

lukeslaboratory, 2019-04-14:
When I think about making things easier for the end user, I ask the question - why have the secret token be the users responsibility? My idea of a better WF goes like this - User signs up on TSD's website with an email address (or local server) -> User installs TSD Plugin on octoprint. User configures octoprint plugin with single input: what's your account email? This triggers your TSD plugin to request a secret key from the server, which can then be used to send an email to user/generate prompt to confirm the request, and start the printer setup. Its similar to when users set up email accounts on their phones, They get an email to confirm the request and the phone does the rest in the background invisible to the user.

This can keep your information-sending the same, but keeps the user out of the addition process, all they need to know is their own email and password, and don't have to pre-generate and copy-paste things.

kennethjiang, 2019-04-15:
Thank you @lukeslaboratory for the detailing your design. It makes a lot of sense except one step: when the plugin to request a secret key from the server for the 1st time, how can the server authenticate the request to make sure it's really from the owner of the provided email address?

An alternative design is for the plugin to generate a UUID, and showed a link that contains this UUID as an HTTP GET parameter for the user to click on. When the user clicks on that link, which point to the server, the server would ask the user to go through the normal sign up/sign in flow to authenticate him/herself. 

This is not how a device would get secret token but I can't think of any holes in this design, which is quite similar to your design when it comes to user experience.

lukeslaboratory, 2019-04-15:
I'm sorry I wasn't clear, but I imagine the request to be a popup when the user next signs in and checks their server page - "There is a printer requesting addition to your account - it identifies itself as "Xxprinterl0rdxX" and maybe provide a snapshot of what it sees. This way the user can verify that it was in-fact their printer, and they've verified it by being logged into their existing account. The UUID is definitely similar, but initiated from the printer side with a direct sign-up link. Either/or would be a better implementation than current, for sure.

kennethjiang, 2021-02-05:
New flow with 6-digit verification code should be a lot better

closed 2021-02-05 by kennethjiang