TheSpaghettiDetective/obico-server #25: windows ml-api build fail

mcmasterp, 2019-03-18:
I was able to start the build under windows in powershell by cd to the TSD directory and then the normal docker-compose up -d, I'm sure there is a syntax that would make it one line but I'm not that versed.

The problem I get is:

ERROR: Service 'ml_api' failed to build: The command '/bin/sh -c wget --quiet -O model/model.weights $(cat model/model.weights.url)' returned a non-zero code: 8

Any ideas? I checked the url in the file and was able to download it manually in a browser so it's reachable from my pc

kennethjiang, 2019-03-18:
Thank you @mcmasterp for reporting this problem to us. Admittedly we have not tested the installation steps on Windows. Sorry about that.

I will work on getting a Windows VM and testing on it. Can you send me the link to the steps for installing docker on Windows? It'll be even better if you can update our `README.md` file with the link and send a PR for it. :) Thanks a lot!

goopsie, 2019-03-18:
I have the same problem, ran `docker-compose up -d`
```PS C:\Users\Goopsie\docc\TheSpaghettiDetective> docker-compose up -d
Building ml_api
Step 1/12 : FROM nvidia/cuda:9.0-devel-ubuntu16.04
 ---> c24bd4961e81
Step 2/12 : RUN apt update
 ---> Using cache
 ---> 5c3c6ea5290f
Step 3/12 : RUN apt install -y libsm6 libxrender1 libfontconfig1 python3-pip vim ffmpeg wget curl
 ---> Using cache
 ---> 874fc000b6e7
Step 4/12 : RUN update-alternatives --install /usr/bin/python python /usr/bin/python3 10
 ---> Using cache
 ---> 09f69feef580
Step 5/12 : RUN update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 10
 ---> Using cache
 ---> 79a451d57b53
Step 6/12 : ADD . /app
 ---> Using cache
 ---> 2f53d931565d
Step 7/12 : WORKDIR /app
 ---> Using cache
 ---> 42c1d1af16a9
Step 8/12 : RUN wget --quiet -O model/model.weights $(cat model/model.weights.url)
 ---> Running in 2281b15ceedf
ERROR: Service 'ml_api' failed to build: The command '/bin/sh -c wget --quiet -O model/model.weights $(cat model/model.weights.url)' returned a non-zero code: 8```

mcmasterp, 2019-03-18:
> Thank you @mcmasterp for reporting this problem to us. Admittedly we have not tested the installation steps on Windows. Sorry about that.
>
> I will work on getting a Windows VM and testing on it. Can [...]

The steps for install were super easy as docker installer for windows comes with docker compose. the other steps are the same as linux with the git lfs and what not

mcmasterp, 2019-03-18:
Well I came up with a solution. I first tried to download the file and point to it locally but that didn't work probably because of file system access issues but I don't actually know. I do some web hosting so I uploaded the file to my server and used that url and it worked. My guess is it's related to the fact that the link was https and mine was http but thats just a guess

kennethjiang, 2019-03-19:
I'm testing #27 on a windows vm and will merge it soon.

kennethjiang, 2019-03-19:
Just merged #27. Thanks @goopsie for your contribution!

@mcmasterp please pull from master and it should work on Windows now.

closed 2019-03-19 by kennethjiang

mcmasterp, 2019-03-19:
> 
> 
> Just merged #27. Thanks @goopsie for your contribution!
> 
> @mcmasterp please pull from master and it should work on Windows now.

yes it works! thank you @goopsie!!

PR #27 (2019-03-19): Fixed #25
Fixes #25

Don't have a linux or mac machine to test on, but should work there as well