TheSpaghettiDetective/obico-server #26: Waiting on connection

r1b4z01d, 2019-03-18:
I have successfully installed the octoprint module and was able to connect and test using the beta servers. Then I wanted to try it locally so, I setup TSD server on my local computer and added the printer. Octoprint verifies the token correctly but it never connects.

mcmasterp, 2019-03-18:
I am also stuck at this point. I verified my port 3334 is allowing connections on the network. the token does verify on the octoprint plugin interface

TomLarrow, 2019-03-18:
Fighting the same issue.

Peeking through the docker logs of the web server I'm seeing a lot of 404's on /ws/dev

GET /ws/dev/ - 404
Not Found: /ws/dev/
GET /ws/dev/

Have tried restarting the containers all at once and individually, and it always has the same message in the logs

luismiparra, 2019-03-18:
Same issue here, perhaps related to run multi instances of octoprint? (Not saying this is the issue just sharing in case other affected has the same multi instances)

kennethjiang, 2019-03-18:
I'll spin up a new VM to see if I can reproduce this error. Stay tuned folks!

kennethjiang, 2019-03-18:
https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/commit/9d938ae6f9a845022d0a8129f211682915f03541 should fix it. 

The cause of this issue was that the previous `docker.yml` was using gunicorn to start `web` server, which ignore the websocket configuration. I just changed it to start `web` using django dev server, which is slightly less efficient but should be fine unless the server is serving tens of printers.

To fix it for your own server, please run:
`cd TheSpaghettiDetective && git pull && docker-compose up --build -d`

Let me know if the fix works for you so that I can close this issue.

TomLarrow, 2019-03-18:
That seems to have fixed it. I now have another issue where the stream to the video camera is broken.  I'll sort through that, but at least the printer is now showing as connected.

mcmasterp, 2019-03-18:
yup that fixed it for me

kennethjiang, 2019-03-18:
Perfect. @TomLarrow Feel free to open another issue for the video stream problem.

closed 2019-03-18 by kennethjiang

r1b4z01d, 2019-03-19:
Worked for me thank!

jeremycruz82, 2021-01-11:
Hello, I tried the above fix and have the issue of Octoprint not being able to connect, however the Token registers successfully. Any ideas?