TheSpaghettiDetective/obico-server #28: Fail gauge in red for over 20 minutes, never failed.

goopsie, 2019-03-19:
TItle says it all, Self hosting the server.
It used to fail whenever it saw spaghetti, but now when there's nothing but spaghetti, it doesn't fail. Even though the fail gauge is in the red the whole time.

kennethjiang, 2019-03-19:
@goopsie do you mean when the printer was connected to TSD cloud, it would fail on sights of spaghetti. But now it doesn't fail with self-hosted server?

mcmasterp, 2019-03-19:
I have this issue I think. I setup gmail smtp as needed (the less secure access instructions are out of date, I had to create an app password, maybe it's because I have two factor suth enabled for google idk)

I tested the system by placing some filament spaghetti on the bed during a print. it was detected and the meter rasied to red but over the 45min print I never got a notice or saw anything occur on the TSD screen. I have it set to alert only, and I tried moving the slider from lowest medium sensitivity setting all the way up to highest high setting but no difference.

goopsie, 2019-03-19:
It would fail when it was connected to the tsd server, and it would fail on my server, but for no reason, it's just stopped failing on my self hosted server.

kennethjiang, 2019-03-19:
We recently added "adaptive detection" to reduce false positives. So basically we no longer look at the absolute p value (probability of spaghetti). Instead the algorithm compares the current EWM mean value to a short-term rolling mean and and a long-term rolling mean. So basically it'll be more likely to alert on a "sudden appearance of spaghetti" than "the spaghetti that's been there from the start". The logic can be found at https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/9d938ae6f9a845022d0a8129f211682915f03541/web/lib/prediction.py#L16

If you run command `docker-compose logs -f web`, you will see lines like this:
`| printer_id: 1 | current_p: 0.3438 | ewm_mean: 0.0688 | rolling_mean_short: 0.1719 | rolling_mean_long: 0.3565 | current_frame_num: 1 | lifetime_frame_num: 2793 |`
You can use these output to determine if TSD is not alerting because the rolling means (short and/or long) "overwhelm" the algorithm.

goopsie, 2019-03-19:
Thanks, That'd explain it!

closed 2019-03-19 by goopsie