TheSpaghettiDetective/obico-server #29: Testing detection and smtp

mcmasterp, 2019-03-19:
I'm trying without success to test the detection system. I turned the sensitivity all the up and I'm trying to place a ball of spaghetti on the build plate mid print. It does get detected and sometimes rises to the red meter level but I never get a notification. I have tried moving the ball around as well as removing it and placing it back after a minute. 

Is there a way to trigger a notice for purposes of testing the smtp? Or a recommended way to produce an alert to confirm the system is working?

kennethjiang, 2019-03-19:
@mcmasterp You can lower the alert thresholds to artificially low level by adding these to `docker.yml`:
```
...
  environment:
    EMAIL_HOST:     # -> such as smtp.gmail.com
...
    THRESHOLD_LOW: '0.05'   # <--- this
    THRESHOLD_HIGH: '0.1'    # <--- and this
...
```

After that restart docker cluster by `docker-compose up --build -d`. 

@goopsie You may want to know about this too. :)

mcmasterp, 2019-03-20:
@kennethjiang ok now that I have the server up and running I tried the code you gave above ( I assumed you meant docker-compose.yml) and it definitely causes it to trigger a failure detected easily. Unfortunately I'm not getting any notification. in the web logs I see 

> web_1     | Twilio settings are missing. Ignored send requests

as the only sign that it took an action. Is there a way to debug the smtp such as watch a log? I feel like it must be an smtp config problem but I can't find any errors in the values I put in the yml file

mcmasterp, 2019-03-20:
I tried a different smtp server that I run instead of gmail and when it gets a failure detected I get this error in the web log:

> 
> web_1     | Internal Server Error: /api/octo/pic/
> web_1     | Traceback (most recent call last):
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner
> web_1     |     response = get_response(request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
> web_1     |     response = self.process_exception_by_middleware(e, request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 124, in _get_response
> web_1     |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/views/decorators/csrf.py", line 54, in wrapped_view
> web_1     |     return view_func(*args, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/views/generic/base.py", line 68, in view
> web_1     |     return self.dispatch(request, *args, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 495, in dispatch
> web_1     |     response = self.handle_exception(exc)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 455, in handle_exception
> web_1     |     self.raise_uncaught_exception(exc)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 492, in dispatch
> web_1     |     response = handler(request, *args, **kwargs)
> web_1     |   File "/app/api/octoprint_views.py", line 88, in post
> web_1     |     alert_if_needed(printer)
> web_1     |   File "/app/api/octoprint_views.py", line 37, in alert_if_needed
> web_1     |     send_failure_alert(printer, pause_print)
> web_1     |   File "/app/app/notifications.py", line 13, in send_failure_alert
> web_1     |     send_failure_alert_email(printer, pause_print)
> web_1     |   File "/app/app/notifications.py", line 34, in send_failure_alert_email
> web_1     |     msg.send()
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/mail/message.py", line 291, in send
> web_1     |     return self.get_connection(fail_silently).send_messages([self])
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/mail/backends/smtp.py", line 103, in send_messages
> web_1     |     new_conn_created = self.open()
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/mail/backends/smtp.py", line 63, in open
> web_1     |     self.connection = self.connection_class(self.host, self.port, **connection_params)
> web_1     |   File "/usr/local/lib/python3.6/smtplib.py", line 251, in __init__
> web_1     |     (code, msg) = self.connect(host, port)
> web_1     |   File "/usr/local/lib/python3.6/smtplib.py", line 338, in connect
> web_1     |     (code, msg) = self.getreply()
> web_1     |   File "/usr/local/lib/python3.6/smtplib.py", line 394, in getreply
> web_1     |     raise SMTPServerDisconnected("Connection unexpectedly closed")
> web_1     | smtplib.SMTPServerDisconnected: Connection unexpectedly closed

kennethjiang, 2019-03-20:
I'm wondering if it was caused by TLS. If your smtp server doesn't support TLS, you will need to set env var `EMAIL_USE_TLS` to `False`. @mcmasterp

mcmasterp, 2019-03-20:
no I used the settings for TLS from my smtp server. I'll give a go using a none secure connection to test and report back

mcmasterp, 2019-03-20:
@kennethjiang 

Well Ive had a print going for about an hour now and while it has shown failure detected on the screen and I have the sensitivity turn all the way up and the threshold set as you said to, It won't initiate a notification so I can't test the unsecure smtp settings. I'm not sure what to do a this point

kennethjiang, 2019-03-20:
Sorry that it's been difficult for you to test it @mcmasterp . 

If you just want to test the smtp function and don't mind a bit of messiness in the process, here is one way to do it:

1. Shell into the *web* container:
`docker-compose exec web bash`

2. Run this command **inside the container** you shelled into in previous step:
`python manage.py shell_plus`

3. Run these command **inside the django shell** you just launched in previous step:
```
from app.notifications import send_failure_alert_email
send_failure_alert_email(Printer.objects.first(), False)
```
The last command will return nothing if everything works, and you should receive an email.

Let me know how it goes.

mcmasterp, 2019-03-21:
@kennethjiang 

Ok I did those steps and the last command did indeed return nothing but I never received an email. I checked by spam box and to be sure I didn't misspell my email address in the Django admin area

mcmasterp, 2019-03-21:
@kennethjiang Ok I figured it out. The issue was with my email address in django. when it is first setup the username is root@example.com. I saw on the admin dashboard a section for email addresses and added my real email address to the user root@example.com. My fault was that I assumed the root@example.com was just a username and emails would go to the email address that I added in the admin. But I inspected the logs on my mail server and found the emails failed because they couldn't find root@example.com. I'm not sure if this is the expected behavior or just something not yet altered or maybe news to you lol. I recommend either implementing some sort of field on setup where the user can input their actual email address or adding steps to the readme explaining the change in the admin thats needed.

As always I am grateful for all the effort and help given.

kennethjiang, 2019-03-21:
Yay!

Good point about the confusion around email address in Django admin. Just removed it with https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/commit/093225307b6c8542fc7f90e1dded2e602726b8d7 . :)

We are quite overwhelmed at this moment but hopefully we can get around to #12 soon so that people don't have to wrestle with smtp settings to run TSD server!

closed 2019-03-21 by kennethjiang