TheSpaghettiDetective/obico-server #30: Streaming video broken?

luismiparra, 2019-03-19:
hi,

I was having issues connecting the plugging and this get fixed here
https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/26

Now, it connects and the image appear when the printer is not printing. When you start pringint we have no video, and also no detection.

Let me know if I can attach some logs, or need more information.

(The idea is amazing!)
Thanks!

kennethjiang, 2019-03-19:
@luismiparra Are you running your own server? Based on your description my best guess is there is an issue in `ml_api` container, or the connection between `web` and `ml_api`.

You can run `docker-compose logs -f` to check the console output of both services. It'll be a lot of output but if you can spot some exceptions that may give you a hint on what went wrong. You can also redirect the console output to a file and attach it here and I'll be happy to take a look.

mcmasterp, 2019-03-19:
I have this after pulling the latest version with windows fixes. here's my logs

`web_1     | Internal Server Error: /api/octo/pic/
web_1     | Traceback (most recent call last):
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/connection.py", line 159, in _new_conn
web_1     |     (self._dns_host, self.port), self.timeout, **extra_kw)
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/util/connection.py", line 57, in create_connection
web_1     |     for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
web_1     |   File "/usr/local/lib/python3.6/socket.py", line 745, in getaddrinfo
web_1     |     for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
web_1     | socket.gaierror: [Errno -2] Name or service not known
web_1     |
web_1     | During handling of the above exception, another exception occurred:
web_1     |
web_1     | Traceback (most recent call last):
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 600, in urlopen
web_1     |     chunked=chunked)
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 354, in _make_request
web_1     |     conn.request(method, url, **httplib_request_kw)
web_1     |   File "/usr/local/lib/python3.6/http/client.py", line 1239, in request
web_1     |     self._send_request(method, url, body, headers, encode_chunked)
web_1     |   File "/usr/local/lib/python3.6/http/client.py", line 1285, in _send_request
web_1     |     self.endheaders(body, encode_chunked=encode_chunked)
web_1     |   File "/usr/local/lib/python3.6/http/client.py", line 1234, in endheaders
web_1     |     self._send_output(message_body, encode_chunked=encode_chunked)
web_1     |   File "/usr/local/lib/python3.6/http/client.py", line 1026, in _send_output
web_1     |     self.send(msg)
web_1     |   File "/usr/local/lib/python3.6/http/client.py", line 964, in send
web_1     |     self.connect()
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/connection.py", line 181, in connect
web_1     |     conn = self._new_conn()
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/connection.py", line 168, in _new_conn
web_1     |     self, "Failed to establish a new connection: %s" % e)
web_1     | urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7f24c832aeb8>: Failed to establish a new connection: [Errno -2] Name or service not known
web_1     |
web_1     | During handling of the above exception, another exception occurred:
web_1     |
web_1     | Traceback (most recent call last):
web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/adapters.py", line 449, in send
web_1     |     timeout=timeout
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/connectionpool.py", line 638, in urlopen
web_1     |     _stacktrace=sys.exc_info()[2])
web_1     |   File "/usr/local/lib/python3.6/site-packages/urllib3/util/retry.py", line 398, in increment
web_1     |     raise MaxRetryError(_pool, url, error or ResponseError(cause))
web_1     | urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='ml_api', port=3333): Max retries exceeded with url: /p/?img=http%3A%2F%2Fweb%3A3334%2Fmedia%2Ftsd-pics%2Fraw%2F1%2F1553020635.jpg (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f24c832aeb8>: Failed to establish a new connection: [Errno -2] Name or service not known',))
web_1     |
web_1     | During handling of the above exception, another exception occurred:
web_1     |
web_1     | Traceback (most recent call last):
web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner
web_1     |     response = get_response(request)
web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
web_1     |     response = self.process_exception_by_middleware(e, request)
web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 124, in _get_response
web_1     |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
web_1     |   File "/usr/local/lib/python3.6/site-packages/django/views/decorators/csrf.py", line 54, in wrapped_view
web_1     |     return view_func(*args, **kwargs)
web_1     |   File "/usr/local/lib/python3.6/site-packages/django/views/generic/base.py", line 68, in view
web_1     |     return self.dispatch(request, *args, **kwargs)
web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 495, in dispatch
web_1     |     response = self.handle_exception(exc)
web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 455, in handle_exception
web_1     |     self.raise_uncaught_exception(exc)
web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 492, in dispatch
web_1     |     response = handler(request, *args, **kwargs)
web_1     |   File "/app/api/octoprint_views.py", line 65, in post
web_1     |     req = requests.get(settings.ML_API_HOST + '/p/', params={'img': internal_url}, headers=ml_api_auth_headers(), verify=False)
web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/api.py", line 75, in get
web_1     |     return request('get', url, params=params, **kwargs)
web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/api.py", line 60, in request
[... cut ...]
web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/adapters.py", line 516, in send
web_1     |     raise ConnectionError(e, request=request)
web_1     | requests.exceptions.ConnectionError: HTTPConnectionPool(host='ml_api', port=3333): Max retries exceeded with url: /p/?img=http%3A%2F%2Fweb%3A3334%2Fmedia%2Ftsd-pics%2Fraw%2F1%2F1553020635.jpg (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f24c832aeb8>: Failed to establish a new connection: [Errno -2] Name or service not known',))`

kennethjiang, 2019-03-19:
Here's how I'd debug it:

1. shell into `web` container: `docker-compose exec web bash`
2. (inside the `web` container), run `apt install -y netcat && nc -v ml_api 3333`

Let me know what's the output of the `nc -v ml_api 3333` command.

mcmasterp, 2019-03-19:
>     2\. apt install -y netcat && nc -v ml_api 3333

@kennethjiang i get:

ml_api: forward host lookup failed: Unknown host

kennethjiang, 2019-03-19:
It looks like for whatever reason the docker compose internal DNS didn't kick in for your cluster. Let me research a bit more to see if there is any clue.

kennethjiang, 2019-03-19:
@mcmasterp  What are the output of these 2 commands, respectively?

`docker-compose ps`
`docker network inspect thespaghettidetective_default`

mcmasterp, 2019-03-19:
@kennethjiang 
The system cannot find the path specified.
             Name                           Command                 State                   Ports
---------------------------------------------------------------------------------------------------------------
thespaghettidetective_ml_api_1   sh -c newrelic-admin run-p ...   Restarting
thespaghettidetective_redis_1    docker-entrypoint.sh redis ...   Up           6379/tcp
thespaghettidetective_tasks_1    bash -c celery -A config w ...   Up           3333/tcp
thespaghettidetective_web_1      bash -c python manage.py c ...   Up           3333/tcp, 0.0.0.0:3334->3334/tcp

and

[
    {
        "Name": "thespaghettidetective_default",
        "Id": "448187ce0263a1a4ad276ffb0094fe15523a7639bcd3316bc2a6493562b0c751",
        "Created": "2019-03-19T18:30:39.4307499Z",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": false,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "172.20.0.0/16",
                    "Gateway": "172.20.0.1"
                }
            ]
        },
        "Internal": false,
        "Attachable": true,
        "Ingress": false,
        "ConfigFrom": {
            "Network": ""
        },
        "ConfigOnly": false,
        "Containers": {
            "1e29392eae8f08b6f40810fafb52cb2c66b8d3d76644066f901369480f29db2a": {
                "Name": "thespaghettidetective_web_1",
                "EndpointID": "e778dbd5c67a9e8bccabeb047a7910616ff3312ec2b705c11a37b3147039c5fa",
                "MacAddress": "02:42:ac:14:00:05",
                "IPv4Address": "172.20.0.5/16",
                "IPv6Address": ""
            },
            "f1ac1424a4662121149963be3292456dbda92beaa1a7faf73dc29cccb9bd0f53": {
                "Name": "thespaghettidetective_redis_1",
                "EndpointID": "7a73fd3c9056e6034c3210e1c752f2d9c8a6aeac8c5e7d104bfb75b7dc330c29",
                "MacAddress": "02:42:ac:14:00:02",
                "IPv4Address": "172.20.0.2/16",
                "IPv6Address": ""
            },
            "fd17457638c0ea90a437217a4b62197ee2f6c7c555165ee2909952847159e5e3": {
                "Name": "thespaghettidetective_tasks_1",
                "EndpointID": "2bf1005bba46ec511256505b113dd56f3b4fd33e0c26cebc60e47b1493cea4ee",
                "MacAddress": "02:42:ac:14:00:04",
                "IPv4Address": "172.20.0.4/16",
                "IPv6Address": ""
            }
        },
        "Options": {},
        "Labels": {
            "com.docker.compose.network": "default",
            "com.docker.compose.project": "thespaghettidetective",
            "com.docker.compose.version": "1.23.2"
        }
    }
]

kennethjiang, 2019-03-19:
Somehow `ml_api` is stuck at "restarting" state. Can you do `docker-compose logs ml_api` to check what got it stuck @mcmasterp ?

mcmasterp, 2019-03-19:
@kennethjiang 

> ml_api_1  | layer     filters    size              input                output
> ml_api_1  |    0 conv     32  3 x 3 / 1   416 x 416 x   3   ->   416 x 416 x  32 0.299 BF
> ml_api_1  |    1 max          2 x 2 / 2   416 x 416 x  32   ->   208 x 208 x  32 0.006 BF
> ml_api_1  |    2 conv     64  3 x 3 / 1   208 x 208 x  32   ->   208 x 208 x  64 1.595 BF
> ml_api_1  |    3 max          2 x 2 / 2   208 x 208 x  64   ->   104 x 104 x  64 0.003 BF
> ml_api_1  |    4 conv    128  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x 128 1.595 BF
> ml_api_1  |    5 conv     64  1 x 1 / 1   104 x 104 x 128   ->   104 x 104 x  64 0.177 BF
> ml_api_1  |    6 conv    128  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x 128 1.595 BF
> ml_api_1  |    7 max          2 x 2 / 2   104 x 104 x 128   ->    52 x  52 x 128 0.001 BF
> ml_api_1  |    8 conv    256  3 x 3 / 1    52 x  52 x 128   ->    52 x  52 x 256 1.595 BF
> ml_api_1  |    9 conv    128  1 x 1 / 1    52 x  52 x 256   ->    52 x  52 x 128 0.177 BF
> ml_api_1  |   10 conv    256  3 x 3 / 1    52 x  52 x 128   ->    52 x  52 x 256 1.595 BF
> ml_api_1  |   11 max          2 x 2 / 2    52 x  52 x 256   ->    26 x  26 x 256 0.001 BF
> ml_api_1  |   12 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512 1.595 BF
> ml_api_1  |   13 conv    256  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x 256 0.177 BF
> ml_api_1  |   14 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512 1.595 BF
> ml_api_1  |   15 conv    256  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x 256 0.177 BF
> ml_api_1  |   16 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512 1.595 BF
> ml_api_1  |   17 max          2 x 2 / 2    26 x  26 x 512   ->    13 x  13 x 512 0.000 BF
> ml_api_1  |   18 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024 1.595 BF
> ml_api_1  |   19 conv    512  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 512 0.177 BF
> ml_api_1  |   20 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024 1.595 BF
> ml_api_1  |   21 conv    512  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 512 0.177 BF
> ml_api_1  |   22 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024 1.595 BF
> ml_api_1  |   23 conv   1024  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x1024 3.190 BF
> ml_api_1  |   24 conv   1024  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x1024 3.190 BF
> ml_api_1  |   25 route  16
> ml_api_1  |   26 conv     64  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x  64 0.044 BF
> ml_api_1  |   27 reorg              / 2    26 x  26 x  64   ->    13 x  13 x 256
> ml_api_1  |   28 route  27 24
> ml_api_1  |   29 conv   1024  3 x 3 / 1    13 x  13 x1280   ->    13 x  13 x1024 3.987 BF
> ml_api_1  |   30 conv     30  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x  30 0.010 BF
> ml_api_1  |   31 detection
> ml_api_1  | mask_scale: Using default '1.000000'
> ml_api_1  |  Try to load cfg: /app/model/model.cfg, weights: /app/model/model.weights, clear = 0
> ml_api_1  | Total BFLOPS 29.338
> ml_api_1  | Loading weights from /app/model/model.weights...Done!
> ml_api_1  |  * Serving Flask app "server" (lazy loading)
> ml_api_1  |  * Environment: production
> ml_api_1  |    WARNING: Do not use the development server in a production environment.
> ml_api_1  |    Use a production WSGI server instead.
> ml_api_1  |  * Debug mode: on
> ml_api_1  |  * Running on http://0.0.0.0:3333/ (Press CTRL+C to quit)
> ml_api_1  |  * Restarting with stat
> ml_api_1  |
> ml_api_1  |  seen 64
> ml_api_1  | Loaded - names_list: /app/model/names, classes = 1
> ml_api_1  | Traceback (most recent call last):
> ml_api_1  |   File "server.py", line 40, in <module>
> ml_api_1  |     app.run(host='0.0.0.0', port=3333, threaded=False)
> ml_api_1  |   File "/usr/local/lib/python3.5/dist-packages/flask/app.py", line 943, in run
> ml_api_1  |     run_simple(host, port, self, **options)
> ml_api_1  |   File "/usr/local/lib/python3.5/dist-packages/werkzeug/serving.py", line 988, in run_simple
> ml_api_1  |     run_with_reloader(inner, extra_files, reloader_interval, reloader_type)
> ml_api_1  |   File "/usr/local/lib/python3.5/dist-packages/werkzeug/_reloader.py", line 332, in run_with_reloader
> ml_api_1  |     sys.exit(reloader.restart_with_reloader())
> ml_api_1  |   File "/usr/local/lib/python3.5/dist-packages/werkzeug/_reloader.py", line 176, in restart_with_reloader
> ml_api_1  |     exit_code = subprocess.call(args, env=new_environ, close_fds=False)
> ml_api_1  |   File "/usr/lib/python3.5/subprocess.py", line 557, in call
> ml_api_1  |     with Popen(*popenargs, **kwargs) as p:
> ml_api_1  |   File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
> ml_api_1  |     restore_signals, start_new_session)
> ml_api_1  |   File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
> ml_api_1  |     raise child_exception_type(errno_num, err_msg)
> ml_api_1  | OSError: [Errno 8] Exec format error

kennethjiang, 2019-03-19:
This looks like a super tricky problem. Google search shows that *OSError: [Errno 8] Exec format error* happens when an incompatible binary is trying to run. The only possible file I can think of is the `model.so` but it appears that file was already successfully executed and even successfully loaded the weights file. And the error didn't occur until Flask tried to "relaunch" the server.

I can't reproduce this error anywhere. Either on a Windows VM, my Mac, or a Linux VM... :(

@luismiparra Are you also having the same issue?

kennethjiang, 2019-03-19:
@mcmasterp One possible way to rule out `model.so` as the trouble-maker is to build https://github.com/AlexeyAB/darknet inside the `ml_api` container. `model.so` is just the renamed `darknet.so` if you can build it. But it takes a lot of knowledge on Linux and ML, as well as trial-and-error to get that build. So I can't blame you if you don't want to go that route...

mcmasterp, 2019-03-19:
@kennethjiang yeah I'm barely keeping up with the work to get TSD working lol, I'm definitely not the guy to try and get that built so sorry.

If it helps I didn't have this issue until I git pulled the newest version, the version from yesterday worked as long as I changed the url in the model.weights.url to a weights file I was hosting on a web server I run

kennethjiang, 2019-03-19:
Yes please revert to the version that worked for you!

mcmasterp, 2019-03-19:
@kennethjiang

Im embarrassed to say I don't know how to revert

kennethjiang, 2019-03-19:
Nothing to be embarrassed about. We are also learning as evidenced by how many problems you guys ran into.

You can revert the changes with roughly these commands:
```
git checkout 9d938ae6f9a845022d0a8129f211682915f03541
## Change the url to point to your server again.
```
I'd also try to rebuild the whole docker cluster to have a clean restart with:
```
docker-compose up --build -d
```

Let me know if it works.

mcmasterp, 2019-03-19:
hmmm. my build failed at the same spot that using my own url fixed before. I think ive got something all turned around. Im gonna try from scratch with master. I'll let you know

mcmasterp, 2019-03-19:
well sadly a refresh and build of master, while successful, didn't end with a working setup. just back to the same behavior from the beginning of this issue thread. 

So I reverted using the code you gave above, it built successfully and starts but web log says a similiar error as before:

> web_1     |
> web_1     | During handling of the above exception, another exception occurred:
> web_1     |
> web_1     | Traceback (most recent call last):
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner
> web_1     |     response = get_response(request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
> web_1     |     response = self.process_exception_by_middleware(e, request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 124, in _get_response
> web_1     |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/views/decorators/csrf.py", line 54, in wrapped_view
> web_1     |     return view_func(*args, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/views/generic/base.py", line 68, in view
> web_1     |     return self.dispatch(request, *args, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 495, in dispatch
> web_1     |     response = self.handle_exception(exc)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 455, in handle_exception
> web_1     |     self.raise_uncaught_exception(exc)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/rest_framework/views.py", line 492, in dispatch
> web_1     |     response = handler(request, *args, **kwargs)
> web_1     |   File "/app/api/octoprint_views.py", line 65, in post
> web_1     |     req = requests.get(settings.ML_API_HOST + '/p/', params={'img': internal_url}, headers=ml_api_auth_headers(), verify=False)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/api.py", line 75, in get
> web_1     |     return request('get', url, params=params, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/api.py", line 60, in request
> web_1     |     return session.request(method=method, url=url, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/sessions.py", line 533, in request
> web_1     |     resp = self.send(prep, **send_kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/sessions.py", line 646, in send
> web_1     |     r = adapter.send(request, **kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/requests/adapters.py", line 516, in send
> web_1     |     raise ConnectionError(e, request=request)
> web_1     | requests.exceptions.ConnectionError: HTTPConnectionPool(host='ml_api', port=3333): Max retries exceeded with url: /p/?img=http%3A%2F%2Fweb%3A3334%2Fmedia%2Ftsd-pics%2Fraw%2F1%2F1553035468.jpg (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f4da4631b00>: Failed to establish a new connection: [Errno -2] Name or service not known',))

kennethjiang, 2019-03-20:
Sorry to hear that! I'm not very surprised since this error doesn't seem to be caused by any of those commits since yesterday. My gut says something else has changed but I can't think of anything that could have caused that.

I'm trying to figure out a way so that we can test the `model.so` separately from the rest of the `ml_api` part to isolate the problem. stay tuned.

kennethjiang, 2019-03-20:
@mcmasterp @luismiparra I think I fixed it, although I don't fully understand where the error came from. It was one of the most bizarre problems I have run into.

Please run this command to get it up and running:
`cd TheSpaghettiDetective && git pull &&  docker-compose up --build -d`

If that doesn't solve the problem, you can just trash the `TheSpaghettiDetective` folder, re-clone from the repo, and repeat the steps to run the server. If you want to keep the existing data, you can simply copy `db.sqlite3` from the old `TheSpaghettiDetective/web` folder to the new one **before** you run `docker-compose up --build -d`

Let me know if it solves the problem for you.

mcmasterp, 2019-03-20:
@kennethjiang yup that fixed it. thank you so much

luismiparra, 2019-03-21:
Hello sorry for delay!,
@kennethjiang 
All fixed now after "cd TheSpaghettiDetective && git pull && docker-compose up --build -d"

Thanks!!!

:)

kennethjiang, 2019-03-21:
Yay!

closed 2019-03-21 by kennethjiang