TheSpaghettiDetective/obico-server #31: Computer specs

mcmasterp, 2019-03-21:
I have 16 printers that will soon move to a backyard office I'm hoping to build or possibly even a small rented office space and I would like to eventually have them all monitored by TSD. I'll likely have to build a new pc for the job so I'm trying to figure out how much computing power I'll need.

What specs would you recommend for such a setup? Is it even possible to build one machine to watch 16 printers? Or maybe it's less demanding than I'm picturing? 

How about the specs for most other users who will monitor just one printer? How important is the gpu? I see watching one printer a lot of cpu usage but not so much on my nvidia gtx1060.

kennethjiang, 2019-03-21:
We haven't done a lot of benchmarking on compute power. I know that just a few printers can grind a very beefy CPU to a halt so you probably do need a GPU. Currently we are using 1 GPU VM (Tesla P100 GPU) that supports 100+ printers just fine. I imagine gtx1060 is at least on-par with P100 or even more powerful so you should be totally fine with it.

mcmasterp, 2019-03-21:
Thanks! How do I go about forcing the load onto the gpu rather than the cpu?

kennethjiang, 2019-03-21:
Change this line: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docker-compose.yml#L44
to 
`HAS_GPU: 'False'`

But I won't be shocked if it doesn't immediately work for you. I remember struggling to make the container in the cloud to use GPU. Let me know what problems you run into and we'll see if/how much I can help.

mcmasterp, 2019-03-22:
@kennethjiang 
It sadly does not work for me because I am running windows and it looks like docker-nvidia is the only solution which is linux only. I guess if I want to do it I'll need to build out a new pc. I guess It doesn't have to be too high spec'd, a miniatx motherboard, ryzen 3, 8gb ram, 128gb ssd, nad a nvidia gtx 1050 2GB seems like it would handle 16 printers easily based on what you said, right?

kennethjiang, 2019-03-22:
@mcmasterp Sorry for the delay. 

I did a quick search and found https://www.nvidia.com/Download/index.aspx?lang=en-us. But I'm not sure if it'd actually work since we haven't tested GPU on windows. 

The new PC spec you mentioned should be good enough. The only (remote) concern is if 2GB GPU memory is big enough. Our current model takes less than 1GB memory, but there is a chance that we will change our model architecture and make it even bigger, or more likely, figure out a way to run prediction in parallel and hence increase the memory demand. Since you can't expand your GPU memory (afaik), you may want to consider that. But I don't want to advise you to over-provision your server, so make the call for yourself.

Is it an option to re-install your current PC to boot into Linux?

mcmasterp, 2019-03-22:
@kennethjiang 

thanks. My current pc does to many other tasks to switch to linux easily. I just got a gtx 1060 6GB off a local facebook seller cheap and ordered but the other bits on amazon so the die is cast lol. any recommendation or linux distro?

kennethjiang, 2019-03-23:
If you have no preferences I guess Ubuntu will be slightly better than other distros since Ubuntu is the platform we test the most.

mcmasterp, 2019-03-25:
@kennethjiang 
OK I have a new machine setup with ubuntu and TSD. I also installed latest nvidia drivers, CUDA, and nvidia-docker. I edited the yml file to have the gpu true statement. How do I determine if TSD is using the GPU?

kennethjiang, 2019-03-25:
- run `nvidia-smi` inside `ml_api` container (not the host) and you may see some % on GPU and a `python3` process. The reason why I said *may* is because I have found it not work this way on some machines, especially when there is not a lot of load.
[image]

- Your CPU % should be quite low. When GPU is not used, you'd see spikes of high CPU %, when API is called to do prediction. But when GPU is used, you have very steady low CPU %.
[image]

mcmasterp, 2019-03-25:
@kennethjiang my apologies for my lack of knowledge but how do i run it inside the ml_api?

kennethjiang, 2019-03-25:
Sorry I forgot to mention it!

`cd TheSpaghettiDetective && docker-compose exec ml_api bash`

mcmasterp, 2019-03-25:
@kennethjiang 
hmmm. when I run 

> sudo docker-compose exec ml_api bash

I get 

>error response from daemon: container is restarting, wait until the container is running

and then ml_api is stuck in restarting and I have to do a docker-compose down and back up to get it working again. I confirmed with 

> docker-compose ps

 that it is reliably up until I run the exec code

kennethjiang, 2019-03-25:
So every time you run `sudo docker-compose exec ml_api bash` and the container gets stuck? 

If so, can you run `sudo docker-compose logs ml_api` to check its console output?

mcmasterp, 2019-03-25:
@kennethjiang 

> ml_api_1  | CUDA Error: CUDA driver version is insufficient for CUDA runtime version
> ml_api_1  | layer     filters    size              input                output
> ml_api_1  |    0 python: ./src/cuda.c:36: check_error: Assertion `0' failed.
> ml_api_1  |  Try to load cfg: /app/model/model.cfg, weights: /app/model/model.weights, clear = 0

kennethjiang, 2019-03-25:
Shit I think I dealt with this before but I don't exactly remember how I solved it...

kennethjiang, 2019-03-25:
@mcmasterp By going through my google history, I recalled a bit more details about this error. Google told me 2 possible causes for this error. One of them was the real cause (for me) the other was not but I don't remember which. :(

1. Cuda driver versions mismatch between host and container.
2. Cuda driver completely missing from either host or container. 

Not sure how helpful this info would be to you... :(

mcmasterp, 2019-03-26:
Yeah I tried all the google-fu I could with no luck. I don't really understand linux or docker containers so it's hard to be able to troubleshoot.

kennethjiang, 2019-03-26:
Is it possible that you somehow let me ssh to your server and poke around? temporary port-forward or something. I'll give you my ssh pub key @mcmasterp

mcmasterp, 2019-03-26:
@kennethjiang absolutely, how can I contact you privately?

kennethjiang, 2019-03-26:
@mcmasterp  my email `k@thespaghettidetective.com`

kennethjiang, 2019-03-27:
Thank you @mcmasterp for letting me use your server. Figured it out. It has to do with https://github.com/NVIDIA/nvidia-docker. Will document the fix. Thanks again for helping other people who also want to run it on a GPU machine (who wouldn't?)

mcmasterp, 2019-03-27:
@kennethjiang

No way thank you! We all owe you guys big time. Maybe you could setup a patreon for those of us who are going to self host but want to support development?

Did you implement the fix on my server or should I follow the fix steps once documented?

kennethjiang, 2019-03-27:
Your server is good to go. :)

Can you take a look at https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/faq.md and verify that the link for "Install Cuda driver" is what you followed? If you did something else to get cuda install on your server, please update the doc and I'll merge it.

lukeslaboratory, 2019-04-14:
Interjecting just to put this in there for relative processing power between the GPU's - 

> We haven't done a lot of benchmarking on compute power. I know that just a few printers can grind a very beefy CPU to a halt so you probably do need a GPU. Currently we are using 1 GPU VM (Tesla [...]

I wouldn't bet on that - depending on what kind of calculations(commands? processes?) you're using, the 1060 could be woefully underpowered - In order of Double->Single->Half performance, the P100 has 4.7teraFLOPS, 9.3teraFLOPS, and 18.7 teraFLOPS of compute performance. the 1060 has 136.7**giga**FLOPS, 4.375teraFLOPS, and 68.36**giga**FLOPS respectively. The stats for the P100 are also in the non-NVIDIA specific configuration, and can increase to 5.3, 10.6, and 21.2teraFLOPS if put in NVLink-Optimized servers.

The memory transfer rate is another big gap - 192.2GB/s at standard clocks for the 1060, but 732GB/s for the p100.

If the commands are single precision, the GTX1060 is 1/2 of the P100, which, if non-memory bounded, would be plenty for up to 50 printers. However - if performance is needed in half or double precision calculations, or is memory speed bound, there might be limitations.

kennethjiang, 2019-04-14:
Wow these are good numbers to know! We are actually quite impressed by the power of P100. Since #4 we have been able to push well beyond 500 printers for one P100 VM.

kennethjiang, 2019-12-19:
`README.md` has been updated with minimum hardware requirements.

closed 2019-12-19 by kennethjiang