TheSpaghettiDetective/obico-server #32: Use Free-mobile home automation notification.

andrelec1, 2019-03-22:
I don't know if is the same in every country, but in France, with the mobile operator "Free", we have a api for sending SMS on t

kennethjiang, 2019-03-22:
That's awesome!

We are a bit overwhelmed at this moment. Will appreciate it if you (or someone else) can send a PR for this.

andrelec1, 2019-03-22:
i realy want to help but sorry i'm not a python user :/
[troll] I'm totaly lost, i can see any semi-colon 😨  [/troll]...

kennethjiang, 2019-03-23:
And no curly braces or even parentheses. Yes Python is nut!

robweber, 2019-04-02:
I see Octoprint has a plugin for this. I'm sure a lot of the code could be re-used to use within this project too: https://github.com/Pinaute/OctoPrint_FreeMobile-Notifier 

Just as a general comment regarding notifications would it be a better process to pull out the notifications area into more of a modular system? I can see different types of notifications being high request items on this type of project in the future. Different email, SMS, MQTT, mobile push services, etc.  It's unfortunate that within Octoprint there isn't a way within the plugin architecture to utilize the existing notification methods there to dispatch the events. Not all notification plugins send things on failure though, many are just on print complete.

closed 2022-05-14 by kennethjiang