TheSpaghettiDetective/obico-server #34: Never seeing octo/status view POST

wesparish, 2019-03-22:
I'm attempting to run my own server (not through docker-compose, but rather a custom K8s chart).

I am seeing POSTs to /octo/pic and I am able to see the image of my printer on the web UI, however, the web UI never shows the filename or estimated time left. The web container logs don't ever indicate any POSTs to /octo/status.

Any ideas on what to check next?

    $ kubectl -n tsd logs the-spaghetti-detective-568cb69cc6-5b295 -c the-spaghetti-detective-web | grep -i 'POST /api/octo/pic' | wc -l`
    16
    $ kubectl -n tsd logs the-spaghetti-detective-568cb69cc6-5b295 -c the-spaghetti-detective-web | grep -i 'POST /api/octo/status' | wc -l
    0

wesparish, 2019-03-22:
It looks like uninstalling and reinstalling the Octoprint plugin resolved the issue. Likely something worth looking in to?

To get into this state, I attempted to change servers and put in a new token for the new server.

closed 2019-03-22 by wesparish

kennethjiang, 2019-03-23:
Plugin version 0.4.0 and later switched to websocket to send the info originally sent to `/octo/status`. Not a lot of users are using 0.3.0 any more. But thank you for reporting this to us! @wesparish

wesparish, 2019-03-23:
Ok, didn't realize that, I'm still very new to the project, but it's very interesting so far!

Is there anything I need to do to get the automatic pause to work? By putting some "pre made spaghetti" on the print bed with the printer in octoprint test mode, the speedometer gauge was in the red for a long time, but never seemed to send a pause command.

kennethjiang, 2019-03-23:
To get the automatic pause to work, change these 2 lines: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/8586a537279f59e59d16350c410ef396b2d440de/web/api/octoprint_views.py#L33

Hopefully the comments are self-explanatory.