TheSpaghettiDetective/obico-server #35: Compose failed

tibmeister, 2019-03-23:
When running docker-compose up -d, here's what I get

`ERROR: Service 'ml_api' failed to build: OCI runtime create failed: container_linux.go:348: starting container process caused "process_linux.go:297: copying bootstrap data to pipe caused \"write init-p: broken pipe\"": unknown`

Docker host is ubuntu 16.04, running Docker CE 18.06.3, Compose version 1.23.2

kennethjiang, 2019-03-23:
I found this issue: https://github.com/docker/for-linux/issues/591 . Not sure if it's the same root cause as the problem you ran into.

We are quite overwhelmed right now. When things settle a bit I'll try to reproduce your environment to see if I can reproduce this error.

tibmeister, 2019-03-23:
Downgrading to Docker CE 18.06.1 seems to be the fix.

kennethjiang, 2019-03-25:
Closing this issue. If you think we should do something about it please re-open it @tibmeister

closed 2019-03-25 by kennethjiang