TheSpaghettiDetective/obico-server #38: Explain minimum requirements

arthurlutz, 2019-03-24:
First off kudos for a great looking project :tada: 

It would be nice to have some indication of the hardware needed to run this server side bit. I believe CUDA only runs on specific hardware, and seeing the nvidia docker image, I guess you need an NVIDIA card. 

Am a bit of a newbie on the subject, but would this project run on a low-end Nvidia Jetson Nano Developer Kit (99$) https://www.sparkfun.com/products/15297 

I think it might be a great project for tinkerers (into 3D printing) that want to look into machine learning and image processing. Thanks for sharing :open_hands: :clap:

kennethjiang, 2019-03-25:
Yes we will add documents for the minimum server requirements (once we figure that out ourselves).

Long story short - you don't even need a GPU server to run it. A PC, even an old one, will work just fine unless you want to support more than a handful of printers.

pattonwebz, 2019-03-25:
I too am interested in these requirements. Bandwidth expectations outgoing from the printer is something I'm keen to know.

I'm planning to run only a single printer and would like to host this in an existing docker swarm. I expect that once the training has been run that the actual CPU requirements are going to be fairly small?

I'm also interested in running it on an addition RPi (I see the documentation text saying that RPi wouldn't be powerful enough to run the learning models... but I actually don't mind waiting several days ;) )

kennethjiang, 2019-03-25:
What I meant in the previous comment was about predicting, not training. There is no way to train current model without a GPU.

I don't think there is a chance to run the server on a Pi without substantially simplifying the model (see #21). But we will love to see someone figure this out because not requiring a PC would remove a big hurtle for people who want to adopt this project.

lukeslaboratory, 2019-04-14:
Could this run on a more powerful variant of a Pi? I have an XU4 which is a big.LITTLE ARM octocore. [https://www.samsung.com/semiconductor/minisite/exynos/products/mobileprocessor/exynos-5-octa-5422/](url) I understand that there aren't many more users that would have this versus a full-blown PC floating around, but there are more powerful little boards out there.

pattonwebz, 2019-04-14:
> What I meant in the previous comment was about predicting, not training. There is no way to train current model without a GPU.
>
> I don't think there is a chance to run the server on a Pi without [...]

I understand now - existing learning models require things that a humble Pi cannot do. Thanks for the reply, sorry I missed it before now.

I am still quite interested in system requirements once people have tested and reported those. I have not yet tried running the server component on my own (client side is running good though and does indeed spot possible failures so good job!).

kennethjiang, 2019-04-14:
@lukeslaboratory Good point about these not-so-popular but powerful single board computers. 

Since in our testing a mediocre PC has way more CPU juice than what's needed to support 1 printer, I'm hopeful that these single board computers, although less powerful than a full-blown PC, will be good enough. Will love to see how people get creative to leverage these hardwares.

@pattonwebz Great to hear you are getting some good results with TSD. :)

lukeslaboratory, 2019-04-14:
@kennethjiang if you ever feel like writing something that could run on a core or two of one of these boards let me know and i'll throw it on my existing NAS that is running on an XU4. I also have an odroid C2 (quad-core 64bit) that I can definitely devote to full-time testing.
[image]
[image]

pattonwebz, 2019-04-14:
@kennethjiang that looks cool, did you design and print that XU4 case yourself? I had a friend that was looking for something along those lines but never quite found what he needed on amazon lol

kennethjiang, 2019-04-14:
@pattonwebz you gave credits to the wrong person. ;)

lukeslaboratory, 2019-04-14:
@pattonwebz - Feel free to give this print a spin. [https://www.thingiverse.com/thing:3563809](url)
I'm the creator.

pattonwebz, 2019-04-14:
Ha! apologies for pinging the wrong person (and polluting this thread). I blame lack of sleep and long work days XD

I sent the case link to my friend, thanks for sharing that Luke :)

andrelec1, 2019-05-06:
I just came out of a meeting at work where we talked about the use of a nvidia jetson for an internal IOT project ...

I directly think  to replace my Raspberry by one of these little beast to control my printer, and why not turn TSD directly into it  ...

Some One have try ?

vr6syncro, 2019-06-30:
anything new from here? have a odroid c2 or rock64 to test.

pattonwebz, 2019-06-30:
I assume the release of the pi4 with the faster CPU and 4GB memory still is not enough to run the training models?

kennethjiang, 2019-07-01:
I'm pretty sure Pi4 is still not enough. For other single-board computers, especially the ones with a beefy GPU, you will have to try to be sure.

kennethjiang, 2019-12-19:
Closing it as `README.md` has been updated with minimum hardware requirements. For anyone who is using single board computer like Jetson Nano, our Discord chat room https://discord.gg/NcZkQfj is a good place to get help/share info.

closed 2019-12-19 by kennethjiang