TheSpaghettiDetective/obico-server #40: Settings for disable recording all time-lapses or time-lapses shorter than certain duration.

kennethjiang, 2019-03-25:


gyro2death, 2019-03-25:
Thank you for opening this issue for me. 

I'd suggest setting it at a default of about 10 minutes. Usually, quick test prints that are done repeatedly are below this length and very few people will need to see their time-lapses as they're almost always about checking the print as it's going on to either manually adjust it live in the print or cancel to try another setting. 

It could also save you guys on processing power to not have to deal with them as well, and could even be actively checked via the ETA from Octoprint and dynamically ignored entirely which may save resources for not having to process the stream through the ML. I can also imagine these shorter test prints probably wouldn't contribute as well to identifying possible failed print defects other than poor first layer adhesion.

kennethjiang, 2019-05-20:
Resolved in master now.

closed 2019-05-20 by kennethjiang