TheSpaghettiDetective/obico-server #41: run on boot

mcmasterp, 2019-03-25:
On a ubuntu system how do we set it to run at boot?

kennethjiang, 2019-03-25:
Good issue! Hopefully someone who is good at it can help us wrap this thing into a service?

mcmasterp, 2019-03-27:
I found this info:

> If your docker.service enabled on system startup
> 
> $ sudo systemctl enable docker
> 
> and your services in your docker-compose.yml has
> 
> restart: always
> 
> all of the services run when you reboot your system if you run below command only once
> 
> docker-compose up -d

It seems to work for me but Ive only tested it a bit

kennethjiang, 2019-03-28:
It sounds like an elegant solution!

kennethjiang, 2019-03-30:
I added `restart: unless-stopped` to all services and tested it on my Mac.

Help is still needed to test it on Linux and Windows.

mcmasterp, 2019-03-30:
I set mine to always since the machine is going to be dedicated to this and I was worried that on a restart it might interpret the system stopping it as a unless-stopped and not restarting it on reboot

lukeslaboratory, 2019-04-14:
@kennethjiang Mine runs on boot on my windows install with docker.

kennethjiang, 2019-04-14:
Thank you @lukeslaboratory for confirming this. It works on my Mac too. I will close this ticket. @mcmasterp please pull the latest code and all containers now should start on system reboot. Feel free to re-open this issue if it doesn't work for you.

closed 2019-04-14 by kennethjiang