TheSpaghettiDetective/obico-server #42: Suggestions for a faster build process

ntova, 2019-03-26:
Hi, I just stumbled upon your project, it looks really cool.
In the readme you mentioned the long build process. I just skimmed over your docker configuration and have a few suggestions:

- [ ] **ml_api**: Use `nvidia/cuda:9.0-runtime-ubuntu16.04` instead of `nvidia/cuda:9.0-devel-ubuntu16.04`, it should be sufficient and uses 1GB less space. -> Saves download time
- [ ] **web**: Use `python:3.6-alpine` instead of `python:3.6` it uses 850MB less space, vim and ffmpeg are available in alpine, vim can probably be removed. -> Saves download time
- [ ] **web**: Make better use of docker caching: First add the requirements file, then pip install, then add the rest of the project. Using this the build process is much faster if no package requirements have changed. -> Faster development cycle
- [ ] **redis** (minor): Use `redis:alpine`, uses 50 MB less space
- [ ] **all docker images**: build them and upload them to Dockerhub. Most of the time the build process downloads and installs apt and pip packages. By prebuilding and uploading to Dockerhub, this can be consolidated into one docker pull for each image. If you still provide the Dockerfiles, anybody can build the images themselves when they want to debug something. -> Saves build time.

By using all suggestions above, you could get rid of all package installations (using the prebuilt image) which saves a lot of time and also save 1.9GB of Downloads.

kennethjiang, 2019-03-26:
Thank you so much @ntova for having started to contribute already!

All your bullet points make perfect sense. If you want to take a stab at it to create a PR, I can give you write access to our docker hub account. Just give me your docker user name and I will add you.

wwsean08, 2019-04-21:
I'm going to run some tests before opening a PR, I've implemented all this and a few other things in the Dockerfiles which should reduce their size and speed up rebuilds while working on the code.

closed 2019-04-23 by kennethjiang

ForsakenHarmony, 2021-05-28:
Any chance you'll put prebuilt images on dockerhub/github container registry?

kennethjiang, 2021-05-29:
Yes there are at: https://hub.docker.com/u/thespaghettidetective

PR #56 (2019-04-23): Reduce docker image sizes and update docker compose to handle alpine images
fixes #42

# Overview of the changes
* ml_api now uses 9.0-runtime-ubuntu16.04 insead of 9.0-devel-ubuntu16.04
* ml_api no longer installs recommended packages (which are not requirements) and explicitly add a few packages because of this
* web now is based off of alpine
* combined a few run statements to reduce the layer count
* rearranged statements so things that don't change often (like requirements.txt or the model weights) are cached on rebuilds
* redis is now using the 5.0-alpine image

I've done some basic test [... cut]
commit 869b2287 (2019-04-23): Reduce docker image sizes and update docker compose to handle alpine images fixes #42
* ml_api now uses 9.0-runtime-ubuntu16.04 insead of 9.0-devel-ubuntu16.04
* ml_api no longer installs recommended packages (which are not requirements) and explicitly add a few packages because of this
* web now is based off of alpine
* combined a few run statements to reduce the layer count
* rearranged statements so things that don't change often (like requirements.txt or the model weights) are cached on rebuilds
* redis is now using the 5.0-alpine image