TheSpaghettiDetective/obico-server #43: image in email broken

mcmasterp, 2019-03-27:
Not sure if it's my configuration that's to blame but when I receive the email saying my print might be failing (I have it set to just notify) the snapshot in the email is not showing

kennethjiang, 2019-03-28:
Sorry for the delay. Have been fighting some fires now that we have more printers on the network and put a lot of pressure on our servers.

Can you forward that email to me? I suspect that the link to the image probably had a wrong host name. @mcmasterp

kennethjiang, 2019-03-28:
Found the problem after checking the email you forwarded to me.

Gmail changed the image link so that the image is "proxied" through their server. So the image in the email is linked to:

https://ci4.googleusercontent.com/proxy/NhTC7UWu0z4iQAfcXHuZby6JJGXdis0uP7MP1HN4pSSc2XKjiGZCBSMshkZrds6d5Lq7v0s3RcgQz9rfFHJk4cTeIvqGy_yIYU8UDSY5IJ5C_Ts=s0-d-e1-ft#http://192.168.1.219:3334/media/tsd-pics/tagged/1/1553720770.jpg

not 

http://192.168.1.219:3334/media/tsd-pics/tagged/1/1553720770.jpg

Since gmail server can't access the image at address `http://192.168.1.219:3334`, the image is now broken in the email.

Not sure if there is a fix for this issue. But it'll definitely go to `FAQ.md`.

mcmasterp, 2019-03-28:
It seems like the only answer would be to somehow attach the image to the email rather than embed it. Is that possible?

kennethjiang, 2019-03-29:
I'm pretty sure sending image as an attachment is doable. The only downside is that the email will be less user-friendly, as we can't weave the image into the text to make it easy for readers to follow.

But I'm open to this suggestion.

mcmasterp, 2019-03-29:
Maybe it would only switch to the attached method if the user has checked off the box in the plugin for using there own server? That way people who use the official service could have the nicely embedded version

kennethjiang, 2019-03-30:
What you said sounds like a reasonable trade-off @mcmasterp .

goopsie, 2019-04-15:
Why not upload the image to imgur or another image hosting website, and link the email to that?

kennethjiang, 2019-04-15:
@goopsie Interesting idea. I doubt that imgur or other similar service would have an api that allows unlimited image uploads from one single api key but I could be wrong. Also is it possible to have the pictures not public/searchable as I am not sure if every user will be ok with having their 3d printing photos public by default.

goopsie, 2019-04-15:
1250 a day, You can privatize at upload, If it's only for self-hosting and when you're sending emails, you'd never hit the limit.

goopsie, 2019-04-30:
You can also portforward, and make the link in the django admin page point to your external ip

closed 2019-05-30 by kennethjiang

mcmasterp, 2019-05-30:
Was this solved? what is the solution?

kennethjiang, 2019-05-30:
The solution is to attach the image to the email. Wasn't hard to do in Django

commit 1c0f89af (2019-05-30): Closes #43.