TheSpaghettiDetective/obico-server #46: Multiple emails/phones for notification

kennethjiang, 2019-03-29:


kennethjiang, 2019-03-29:
An alternative approach is to attach email/phone to printers, as opposed to user account.

wwsean08, 2019-05-10:
This may actually be possible, looking at allauth briefly it does seem like it supports a user having multiple email addresses though I'll try it out and see if that's the case

kennethjiang, 2019-05-10:
Good catch. I believe I have seen that too.

wwsean08, 2019-05-11:
Yeah I tested last night and it 100% does support it, I was able to get more than one email attached to my account, I'm going to work on getting SMTP working (for some reason i'm getting stack traces when pointing to my email provider, i probably have a wrong setting somewhere) and will validate that it sends to all the email accounts on my...account.

wwsean08, 2019-05-11:
All right after doing some tests I've determined the following:

1. An account can have one or more email addresses associated with it right now with no code changes (though some CSS would be nice to make the UI a little more usable).
2. Only the one which is marked as the "primary" email address will get notified on a failure currently.  A code change would need to be made in order to email all the (verified) addresses associated with the account.

kennethjiang, 2019-05-12:
Thank you so much @wwsean08 for taking a deep dive here.

kennethjiang, 2019-06-04:
closed by #72

closed 2019-06-04 by kennethjiang