TheSpaghettiDetective/obico-server #47: Add "Do not pause on failure for current print" toggle to My Printer page

kennethjiang, 2019-04-02:
When user observes false positives or otherwise don't want the print to be paused on possible failures, and they don't want to change the settings since they are afraid they will forget to change it back. This toggle that's effective only for current print is a convenient way to temporarily disable pause-on-print.

lukeslaboratory, 2019-04-14:
Further Proposal - Add a feature that could have custom Gcode to disable/enable monitoring/pause functionality on a per-print basis.

Example - Insert command M69420 S0 for Full Monitoring/Pause, M69420 S1 for Monitor but no pause, S2 for No monitor no pause. This could be REALLY useful for hobby users with the lower-teir hour-limited plan - No extra fiddling other than a line in the file that could be extra vulnerable.

kennethjiang, 2019-04-14:
Can you elaborate a bit more on the use cases you have in mind for supporting these custom gcode @lukeslaboratory ? Since I'm not personally familiar with them, I'm wondering under what circumstances will users find it useful to send these gcodes.

lukeslaboratory, 2019-04-14:
The lion that can have its hair molded post-printing is the most popular print that I could see users wanting to disable monitoring on it it would be nice if that could be included on a per-file basis.

I tend to adjust settings on a per-print basis - I'd use some sort of scripting in cura/Simplify3d/PathIO to add that in as my starting G-code if I didn't want to be notified if it was failing. If I was setting it up for a print in Nylon, CarbonFiber filled Polycarbonate, I'd want a way in the Gcode to definitely make it so it was monitored and paused to ensure no loss of expensive filament. 

Having it in the G-code allows the user to add it to their pre-print scripting.

Another option would be to enable a toggle that could be clicked on, but ALSO enabled/disabled with the G-code from the file - I could default have it to "monitor but no pause" but with that extra G-code/tag in the file I could have it automatically flip to "monitor and pause". This way in slicers it could automatically be enabled with troublesome filaments like Ninjaflex which has a habit of not sticking to beds.
[image]

kennethjiang, 2019-04-14:
Ah I understand what you mean now. This won't be too difficult to implement but I want to think about the usability of this feature.

kennethjiang, 2019-05-20:
Added a toggle to temporarily turn off alerting for current print: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/commit/253e7d1550b3711e939ac0b6b86bd40772b8e9bb

@lukeslaboratory What you proposed would take quite a bit of effort to implement and document for most users to be able to do correctly. I'm closing this issue for now. If you feel strongly about your proposal, please open another issue for us to discuss in details.

closed 2019-05-20 by kennethjiang