TheSpaghettiDetective/obico-server #49: Don't delete timelapses on image rebuild

EleRas, 2019-04-04:
Currently a docker-compose up --build results in an explicit deletion of the timelapses while rebuilding the web - container:

> Deleting 'media/tsd-timelapses/private/1_p.json'
> Deleting 'media/tsd-timelapses/private/2.mp4'
> Deleting 'media/tsd-timelapses/private/1_tagged.mp4'
> Deleting 'media/tsd-timelapses/private/1.mp4'
> Deleting 'media/tsd-timelapses/private/1_poster.jpg'
> Deleting 'media/tsd-timelapses/private/2_poster.jpg'
> Deleting 'media/tsd-timelapses/private/2_p.json'
> Deleting 'media/tsd-timelapses/private/2_tagged.mp4'

It didn't seem to matter if this directory was mounted as an external volume or not. This does seem counter-productive if you want to keep the timelapses for future training.

kennethjiang, 2019-04-04:
Based on my understanding, these output was from the command `python manage.py --collect-statics` in the process of building a docker image. Since the folder `media` will be remounted when the container really runs, the folders and files being deleted were actually just copies of the original files. 

Although I'm quite sure about my understand but  please confirm whether your timelapses were really deleted or not @EleRas

EleRas, 2019-04-04:
I found my mistake: I mounted the external volume to /media, not /app/static_build/media - so the data wasn't stored in the external volume at all. This should documented in either the docker-compose.yml or the readme ;)

kennethjiang, 2019-04-04:
What do you think needs to be documented? Should we document where to mount the volume or these messages can be ignored?

EleRas, 2019-04-04:
Where to mount the volume, since this isn't documented yet, as far as I can see

kennethjiang, 2019-04-04:
We actually specified the mount point (not `media` folder, but its parent folder) at: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/5664de2b0fe5461fa64bc81b85bd33241eb5bf97/docker-compose.yml#L8

I'm actually curious why you chose to mount `media` at a different directory? Was it because the original mount point (volume mapping in Docker's term) somehow didn't work for you?

EleRas, 2019-04-04:
That was the first thing I did - but this resulted in the media - files being stored in the git - checkout and being deleted during a docker-compose build. Having seen the error message "Deleting 'media..." I thought that the data was stored in /media as well and the data in the git - checkout just being artefacts.

kennethjiang, 2019-04-04:
The `static_build` dir is in `.gitignore`: `https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/5664de2b0fe5461fa64bc81b85bd33241eb5bf97/.gitignore#L126`

Are you sure somehow it ended up in git? Can you elaborate more?

EleRas, 2019-04-04:
It didn't end in git itself, what I meant was: the media was in a directory below the git - checkout (<gitcheckout>/web/static_build) - and <gitcheckout>/web/static_build is being emptied by docker-compose build

kennethjiang, 2019-04-04:
Sorry for misunderstanding you @EleRas ?

If you revert your changes to the mount point (basically going back to the original code by doing `git reset --hard`), will these files (the ones in `TheSpaghettiDetective/web/static_build`) still be deleted after `docker-compse up --build -d`?

EleRas, 2019-04-04:
I currently have a long print running, but as soon as it's finished, I'll try it again.

EleRas, 2019-04-05:
Okay, I could reproduce it reliable:
```
git clone https://github.com/TheSpaghettiDetective/TheSpaghettiDetective.git tsd
cd tsd
docker-compose up --build -d
[configure via web] && [run timelapse]
OR just sudo mkdir web/static_build/media && sudo touch web/static_build/media/test
ls -lisah web/static_build/media/
total 8,0K
97255700 4,0K drwxr-xr-x  2 root root 4,0K Apr  5 19:32 .
96996053 4,0K drwxr-xr-x 12 root root 4,0K Apr  5 19:32 ..
97255701    0 -rw-r--r--  1 root root    0 Apr  5 19:32 test
docker-compose up --build -d
ls -lisah web/static_build/media/
total 8,0K
97255700 4,0K drwxr-xr-x  2 root root 4,0K Apr  5 19:34 .
96996053 4,0K drwxr-xr-x 12 root root 4,0K Apr  5 19:34 ..
```

So the media - directory is emptied during ```docker-compose up --build```, it also doesn't matter if TSD was configured or not. Another interesting point: the web/db.sqlite3 is _not_ deleted during the rebuild.

kennethjiang, 2019-04-06:
Found the problem and fixed it. Thank you @EleRas for reporting this to us!

commit fff196e9 (2019-04-06): closes #49. -c in collectstatic would delete media files too. Disabling it
closed 2019-04-06 by kennethjiang