TheSpaghettiDetective/obico-server #51: Server does not Compile

lukeslaboratory, 2019-04-12:
Performing updates on the server (or fresh installs) with commits past commit 19942c169a98350d79021ba1392537d3cce1a528 results in not being able to connect to the server locally and unable to configure.

kennethjiang, 2019-04-12:
Are TSD pages, or django admin pages loading in browser? If not, can you run command `docker-compose logs --tail=500 web` and post the output here? @lukeslaboratory

lukeslaboratory, 2019-04-13:
Sorry for not including more information before:
When I run a fresh install and try to connect, I get a white page with "Server Error (500)"

Below is the log from running that command.

Attaching to thespaghettidetective_web_1
web_1     |
web_1     | 195 static files copied to '/app/static_build', 601 post-processed.
web_1     | Operations to perform:
web_1     |   Apply all migrations: account, admin, app, auth, contenttypes, sessions, sites, socialaccount
web_1     | Running migrations:
web_1     | /usr/local/lib/python3.6/site-packages/psycopg2/__init__.py:144: UserWarning: The psycopg2 wheel package will be renamed from release 2.8; in order to keep installing from binary please use "pip install psycopg2-binary" instead. For details see: <http://initd.org/psycopg/docs/install.html#binary-install-from-pypi>.
web_1     |   """)
web_1     |   Applying contenttypes.0001_initial... OK
web_1     |   Applying contenttypes.0002_remove_content_type_name... OK
web_1     |   Applying auth.0001_initial... OK
web_1     |   Applying auth.0002_alter_permission_name_max_length... OK
web_1     |   Applying auth.0003_alter_user_email_max_length... OK
web_1     |   Applying auth.0004_alter_user_username_opts... OK
web_1     |   Applying auth.0005_alter_user_last_login_null... OK
web_1     |   Applying auth.0006_require_contenttypes_0002... OK
web_1     |   Applying auth.0007_alter_validators_add_error_messages... OK
web_1     |   Applying auth.0008_alter_user_username_max_length... OK
web_1     |   Applying auth.0009_alter_user_last_name_max_length... OK
web_1     |   Applying app.0001_initial... OK
web_1     |   Applying account.0001_initial... OK
web_1     |   Applying account.0002_email_max_length... OK
web_1     |   Applying admin.0001_initial... OK
web_1     |   Applying admin.0002_logentry_remove_auto_add... OK
web_1     |   Applying admin.0003_logentry_add_action_flag_choices... OK
web_1     |   Applying app.0002_seed_data_2247... OK
web_1     |   Applying app.0003_auto_20190202_1354... OK
web_1     |   Applying app.0004_auto_20190210_1254... OK
web_1     |   Applying app.0005_auto_20190219_2337... OK
web_1     |   Applying app.0006_bed_heater_off_default_false... OK
web_1     |   Applying app.0007_add_phone_num_to_user... OK
web_1     |   Applying app.0008_create_printer_preidction... OK
web_1     |   Applying app.0009_add_detective_sensitivity_to_printer... OK
web_1     |   Applying app.0010_auto_20190303_2125... OK
web_1     |   Applying app.0011_print_prediction_json_url... OK
web_1     |   Applying app.0012_auto_20190323_2103... OK
web_1     |   Applying sessions.0001_initial... OK
web_1     |   Applying sites.0001_initial... OK
web_1     |   Applying sites.0002_alter_domain_unique... OK
web_1     |   Applying socialaccount.0001_initial... OK
web_1     |   Applying socialaccount.0002_token_max_lengths... OK
web_1     |   Applying socialaccount.0003_extra_data_default_dict... OK
web_1     | /usr/local/lib/python3.6/site-packages/psycopg2/__init__.py:144: UserWarning: The psycopg2 wheel package will be renamed from release 2.8; in order to keep installing from binary please use "pip install psycopg2-binary" instead. For details see: <http://initd.org/psycopg/docs/install.html#binary-install-from-pypi>.
web_1     |   """)
web_1     | Exception inside application: 'NoneType' object has no attribute '_wrapped'
web_1     |   File "/usr/local/lib/python3.6/site-packages/channels/sessions.py", line 179, in __call__
web_1     |     return await self.inner(receive, self.send)
web_1     |   File "/usr/local/lib/python3.6/site-packages/channels/middleware.py", line 40, in coroutine_call
web_1     |     await self.resolve_scope(scope)
web_1     |   File "/usr/local/lib/python3.6/site-packages/channels/auth.py", line 166, in resolve_scope
web_1     |     scope["user"]._wrapped = await get_user(scope)
web_1     |   'NoneType' object has no attribute '_wrapped'
web_1     | GET /admin/
web_1     | {'HTTP_HOST': '192.168.0.139:3334', 'HTTP_CONNECTION': 'keep-alive', 'HTTP_UPGRADE_INSECURE_REQUESTS': '1', 'HTTP_USER_AGENT': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/73.0.3683.103 Safari/537.36', 'HTTP_ACCEPT': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3', 'HTTP_ACCEPT_ENCODING': 'gzip, deflate', 'HTTP_ACCEPT_LANGUAGE': 'en-US,en;q=0.9', 'HTTP_COOKIE': 'csrftoken=UYfQ4RpOsBWNcADsf4pUV0GCeBMIjGE30VvU3kY4QkgyEyXxd6Q5P0Szpu8QHcZU; sessionid=zl0okjeq8znju6a2cxvw6o103h5hoilz'}
web_1     | GET /admin/ - 302
web_1     | GET /admin/login/?next=/admin/
web_1     | {'HTTP_HOST': '192.168.0.139:3334', 'HTTP_CONNECTION': 'keep-alive', 'HTTP_UPGRADE_INSECURE_REQUESTS': '1', 'HTTP_USER_AGENT': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/73.0.3683.103 Safari/537.36', 'HTTP_ACCEPT': 'text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3', 'HTTP_ACCEPT_ENCODING': 'gzip, deflate', 'HTTP_ACCEPT_LANGUAGE': 'en-US,en;q=0.9', 'HTTP_COOKIE': 'csrftoken=UYfQ4RpOsBWNcADsf4pUV0GCeBMIjGE30VvU3kY4QkgyEyXxd6Q5P0Szpu8QHcZU'}
web_1     | Internal Server Error: /admin/login/
web_1     | Traceback (most recent call last):
web_1     |   File "/usr/local/lib/python3.6/site-packages/django/contrib/sites/models.py", line 41, in _get_site_by_request
web_1     |     SITE_CACHE[host] = self.get(domain__iexact=host)
[... cut ...]
web_1     | {'HTTP_HOST': '192.168.0.139:3334', 'HTTP_CONNECTION': 'keep-alive', 'HTTP_USER_AGENT': 'Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/73.0.3683.103 Safari/537.36', 'HTTP_ACCEPT': 'image/webp,image/apng,image/*,*/*;q=0.8', 'HTTP_REFERER': 'http://192.168.0.139:3334/admin/login/?next=/admin/', 'HTTP_ACCEPT_ENCODING': 'gzip, deflate', 'HTTP_ACCEPT_LANGUAGE': 'en-US,en;q=0.9', 'HTTP_COOKIE': 'csrftoken=UYfQ4RpOsBWNcADsf4pUV0GCeBMIjGE30VvU3kY4QkgyEyXxd6Q5P0Szpu8QHcZU'}
web_1     | GET /favicon.ico - 404
web_1     | Not Found: /favicon.ico

kennethjiang, 2019-04-14:
Found the problem. @lukeslaboratory please pull to see if it's fixed. Thanks for reporting this bug!

closed 2019-04-14 by kennethjiang

lukeslaboratory, 2019-04-14:
Confirm Successful Close

commit 6a97dc1d (2019-04-14): closes #51. SITE_ID is still used when no record match hostname is found.