TheSpaghettiDetective/obico-server #52: Add Pushbullet integration

lukeslaboratory, 2019-04-14:
For local implementations of TSD, it would be really nice to add Pushbullet integration for web-only notification. Documentation: [https://docs.pushbullet.com/](url) This would allow home users to still get notifications outside of just email

kennethjiang, 2019-04-14:
Thank you @lukeslaboratory for pointing us to https://docs.pushbullet.com/ . It looks like a nice project. I have put it on our list of "promising things to look/try".

lukeslaboratory, 2019-04-14:
I've had alot of success integrating them into my projects here - I currently use it as a Crypto bot notifier as well as my printer print completion - there's an octoprint plugin already - [https://github.com/OctoPrint/OctoPrint-Pushbullet](url)

kennethjiang, 2019-08-02:
Resolved by #83 . Kudos to @ajdavids for his awesome PR.

closed 2019-08-02 by kennethjiang