TheSpaghettiDetective/obico-server #53: My Printer "Feed" doesn't keep up?

karl0ss, 2019-04-19:
I am running the docker container on my full server, and have octopi running on a pi3, im using the offical raspberry pi camera.

Both are connected via LAN and are communicating, I can see the webcam feed in TSD, but the feed is not real time, is it just taking snapshots rather than real time feed? I felt from the sample videos it was real time, please see the attached video of what I mean/am seeing.

https://youtu.be/bNuAHzJ2JJo

If you see right at the end of the video, is when the frame updates in TSD...

Thanks

kennethjiang, 2019-04-19:
The webcam feed does have significant latency, as [explained here](https://www.thespaghettidetective.com/docs/webcam-feed-is-laggy/).

The sample videos on the website are the time-lapses compiled after the prints are done.

karl0ss, 2019-04-19:
I have just noticed this, the is bl00dy brilliant...

closed 2019-04-19 by karl0ss