TheSpaghettiDetective/obico-server #54: Gyroid infill false alarms

Moharmate, 2019-04-20:
I have found that when printing with gyroid infill it mistakes the squiggly infill pattern as a false flag and constantly pauses the print.

kennethjiang, 2019-04-20:
Thank you @Moharmate for reporting this problem. Based on user's feedback we have found it to be a common problem. We are currently training an improved machine learning model and will make sure we incorporate some pics with Gyroid infill in the data set.

Meanwhile, you can adjust the settings so that the print won't be paused in case of possible failures: https://www.thespaghettidetective.com/docs/detection-print-job-settings/#when-potential-failure-is-detected

Moharmate, 2019-04-20:
Thanks for the info. I have adjusted my settings and ill just stick with cubic infill for now. Love what your working on.

closed 2019-04-20 by Moharmate