TheSpaghettiDetective/obico-server #58: versions

mcmasterp, 2019-04-30:
I want to keep an eye on development progress and update my server when important changes are made. Is there any chance of instituting versions and changelogs so I know my version and can see whats changed?

kennethjiang, 2019-05-09:
Sorry not sure why your issue fell through the cracks. Sorry about that

We believe that the most important task at hand right now is to improve the prediction model, which we are spending a lot of time on. So I feel like versioning the server is almost equivalent to versioning the model, which we will definitely do. But maybe that's not exactly what you mean?

mcmasterp, 2019-05-09:
No worries, glad your busy improving. I meant the whole package but of
course the model is a big part of that. You go ahead and work on the
important stuff and we can circle back around to this. If you wanted to do
anything immediately maybe you could just put a dated changelog at the
bottom of the Readme updating it everytime you make a significant
improvement

kennethjiang, 2019-08-22:
I'm closing this issue as we have been following [Continuous Delivery](https://continuousdelivery.com/) rather than versioned release process. So it's quite tricky to identify "versions". However it's a good point that we need to keep the documents up to date with the code and hopefully we have done a good enough job on that.

closed 2019-08-22 by kennethjiang