TheSpaghettiDetective/obico-server #62: Unable to compose with latest docker release.

nate2014jatc, 2019-05-11:
Hello.
Upon following your tutorial, I am presented with the following error during the `docker-compose` phase.
[image]

nate2014jatc, 2019-05-11:
Changing Ln1 of `docker-compose.yml` from `version: '2.4'` to `version: '2.2'` seems to allow the process to continue.

wwsean08, 2019-05-11:
2.4 seems to work for me, so the version does seem to exist.

wwsean08, 2019-05-11:
just a thought, what versions of docker compose and docker are you running:

```
sesmith@ubuntu:~/git/TheSpaghettiDetective$ docker-compose --version
docker-compose version 1.24.0, build 0aa5906
sesmith@ubuntu:~/git/TheSpaghettiDetective$ docker version
Client:
 Version:           18.09.2
 API version:       1.39
 Go version:        go1.10.4
 Git commit:        6247962
 Built:             Tue Feb 26 23:52:23 2019
 OS/Arch:           linux/amd64
 Experimental:      true

Server:
 Engine:
  Version:          18.09.2
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.10.4
  Git commit:       6247962
  Built:            Wed Feb 13 00:24:14 2019
  OS/Arch:          linux/amd64
  Experimental:     false
```

kennethjiang, 2019-05-12:
@nate2014jatc it looks like your docker is at `1.xx` version based on this chart here: https://docs.docker.com/compose/compose-file/compose-versioning/

nate2014jatc, 2019-05-13:
Sorry for the delay.
```
nate@micro  ~  docker --version
Docker version 18.09.6, build 481bc77
 nate@micro  ~  docker-compose --version
docker-compose version 1.17.1, build unknown
 ```

nate2014jatc, 2019-05-13:
```
nate@micro  ~  docker version
Client:
 Version:           18.09.6
 API version:       1.39
 Go version:        go1.10.8
 Git commit:        481bc77
 Built:             Sat May  4 02:35:57 2019
 OS/Arch:           linux/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          18.09.6
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.10.8
  Git commit:       481bc77
  Built:            Sat May  4 01:59:36 2019
  OS/Arch:          linux/amd64
  Experimental:     false
```

kennethjiang, 2019-05-13:
That must be why. It looks like you need docker compose 1.21.0+ . Since docker-compose is just a simple script, it should be quite easy to upgrade. Let me know if you still experience the issue after upgrading docker-compose.

I also updated the README file for the version requirement. Thank you for catching this for us @nate2014jatc

nate2014jatc, 2019-05-16:
Solved by updating 'docker-compose' manually

kennethjiang, 2019-11-20:
What's the version of your `docker-compose` @arcreigh ?

arcreigh, 2019-11-20:
@kennethjiang Sorry I meant +1 to the fix I should have specified that. Feel free to close this :)

closed 2019-11-20 by kennethjiang