TheSpaghettiDetective/obico-server #63: Send verification email and require verification for notification

wwsean08, 2019-05-11:
TheSpaghettiDoctor should send a verification email to users when they sign up with email/password to verify they actually own the account.  Social login proves they control the account and thus should control the email address associated with it so no need to send a verification in that case.

On top of that if an email address is not verified then it shouldn't get email notifications.  Most likely this will be a breaking change if it gets implemented fully (given the specification of don't notify if the emails not verified), so I'm open to feedback or alternative ideas, though it should be possible to write some code to send a verification email to every user at once as an admin (haven't validated that).

IMO this is a requirement for #46

kennethjiang, 2019-05-12:
Email verification is definitely on our roadmap. But given we have our hands full at this moment, it'll be great if someone can help implement it.

If I remember correctly allauth does support email verification. I believe what need to happen will roughly be:

1. Enable email verification in allauth
2. Customize the email and verification page 
3. Change the email sending code to not send email if not verified
4. Deal with existing user accounts (either mark all of them as "verified" or require all of them to verify).

I can deal with 4 but any help on 1-3 will be appreciated.

wwsean08, 2019-05-12:
Yup I've started working on this, so far have 1 done and 2 coming along
[image]

wwsean08, 2019-05-30:
I know it's 17 days later but this is slowly coming along.  Here are some screenshots so if you think anything should be changed let me know:

[image]
[image]
[image]

kennethjiang, 2019-05-30:
These screenshots look great! I particularly appreciate how you managed to keep the user experience and look & feel consistent with existing pages. Thank you so much Sean!

Do you plan to send alert emails to all email addresses, all verified email addresses, or let user decide which address(es) to send to?

wwsean08, 2019-05-30:
The current plan is to send to all verified addresses with the ability (at the server level) to allow sending to unverified addresses via a setting.  

The main reason I want to go that route is it makes things a-lot simpler as I don't have to manipulate the table allauth uses to store a users emails or create a table that joins that data of user->email mapping for notifications which seems like a bad idea as well.

Currently I need to test my updated email code to make sure that it actually does work now (I coded that up last night but didn't validate it yet) and check out how social accounts are created and make sure this doesn't break them.

Also I plan to put some sort of notification banner up for users if they don't have any verified email address to let them know they won't receive notifications because it would be really bad if someone just planned to get notifications but didn't have any indication of why they aren't getting them.

kennethjiang, 2019-05-31:
You design sounds good to me. Actually I'm huge fan of KISS so I'd probably even prefer just to send emails to verified addresses and do the warning banner for the users who don't have one, rather than bothering with another setting to allow them override this behavior.

kennethjiang, 2019-05-31:
Just saw your comment in the PR and realized I misunderstood what you mean by a setting. A env var setting is a perfect choice. We will probably leave it as `False` by default but then turn it to `True` for the official website.

PR #72 (2019-06-04): Initial implementation to fix #63
This change fixes #63 and makes the following changes:
* Email confirmations are sent when an email server is configured for new users (non-social)
  * Social accounts are verified by default (this isn't a change I made, just calling it out
* By default a user is required to confirm their email address to receive notification emails  **behavior change**
  * The EMAIL_REQUIRES_VERIFICATION allows you to override this behavior
  * It would be trivial to make the old behavior default
* All (verified) email accounts receive notification emails on print failure
* D [... cut]
commit bc7189b7 (2019-06-04): Initial implementation to fix #63
closed 2019-06-04 by kennethjiang