TheSpaghettiDetective/obico-server #64: ERROR - Backing off XXX seconds: Failed to connect to websocket server

goopsie, 2019-05-14:
[octoprint.log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/3179364/octoprint.log)

Not getting anything on the printers page, No webcam feed. The token is correct. Nothing weird in the server logs, except for `web_1     | WebConsumer: Closed websocket with code: None`
whenever i click off of the spaghetti detective page

goopsie, 2019-05-14:
and restarting octoprint fixed it.

kennethjiang, 2019-05-14:
Were you using your own server? The plugin does require a restart when the server url is changed from the default one to your own server. Let me know if it was not what happened and I'd like to get to the bottom of it. @goopsie

goopsie, 2019-05-14:
Yes, it was my own server. Didn't change the url. It happened right after updating Octoprint, that might be the cause? It was working fine before updating

kennethjiang, 2019-05-14:
The log seems to show the connection errors came up right after OctoPrint. And they look like real connection issues (as opposed to some root errors masqueraded as a connection issue). It isn't obvious to me why OctoPrint upgrade would trigger that...

I suspect this problem may come back to you. If this error happens again, please don't restart OctoPrint right away. Instead, ssh to your Pi and run `nc -v your_tsd_server_ip your_tsd_server_port`. By that we can determine if it is a real connection issue or some else messed up in the plugin.

goopsie, 2019-05-14:
Ah, sorry. Will do next time!

kennethjiang, 2019-05-14:
Oh nothing to be sorry about! I never told you that before. :) And thank you for being helpful in the PR and providing a lot of details in the issue.

goopsie, 2019-05-15:
Yeah, i'm having the same problem. Netcat can connect to my server fine
`pi@raspberrypi:~ $ nc -v 192.168.1.186 3334
Connection to 192.168.1.186 3334 port [tcp/*] succeeded!
`

kennethjiang, 2019-05-15:
We didn't realize OctoPrint 1.3.11 broke the plugin until last night. Just pushed out a new release 0.5.4. Please upgrade whenever OctoPrint prompts you to do so and let me know if the problem happens again.

goopsie, 2019-05-18:
Haven't had any problems since the update, I think it's fine now. Thanks a ton!

closed 2019-05-18 by kennethjiang