TheSpaghettiDetective/obico-server #66: More of a question - how to deal with large tsd-pics?

cmsimike, 2019-05-21:
Hi,

I have TSD setup and working great! Thank you for your effort so far with this! I've never been comfy enough to let my 3d printer go overnight but I am slowly building my trust thanks to this tool. The option to run it locally is <3. I love self hosting so I really appreciate the fact I can just run the server myself.

Looking at my install, I noticed a lot of the file system is being taken up by the contents of tsd-pics:
[image]

I tried looking around and don't think I saw anything about what's going on here.
1) What is this dir for?
- I presume it's to store the snapshots of previous and current prints
2) If above is correct, is there a reason to keep the contents? I presume the images there are important to link in an emai, but I'm concerned over how big the dir is growing just after a week. What is best practice to keep this size in check?

kennethjiang, 2019-05-21:
Yes your assumptions were correct. We have been aware about this issue from the very beginning and have this issue #2 for it. But have not got around to it.

Anyone is more than welcome to send a PR for it. :) It could be just a celery task that runs once in a while.

We are using Google cloud as our file storage which comes with its own purging mechanism so it's not been a problem for us even if we are running thousands of printers right now.

cmsimike, 2019-05-22:
Gotcha! I thought i had sufficiently searched the issues for this, sorry about the dupe!
I'll look into the celery task. Would be nice for everyone to have.

Closing this out in favor of #2

closed 2019-05-22 by cmsimike