TheSpaghettiDetective/obico-server #70: Model based detection?

bugilt, 2019-05-30:
Is there any sort of effort to use the model and the bed positioning to align a virtual model for comparison? I would imagine printing a specific model to calibrate the camera alignment. Then getting perspective, scale, lens distortion, ect... Then modeling what the camera should see digitally. Then comparing the two with AI to detect all sorts of things. Just some thoughts I figured I'd share.

kennethjiang, 2019-05-30:
Thank you for sharing your thoughts @bugilt This is not how it's currently done. Current algorithm is based on deep learning, which is similar to how facial recognization works.

I'll keep this issue open to get people's input on how your proposal can be implemented, and/or the possible approaches to automatically detecting failures.

HakuG, 2019-06-04:
You'll need the g-code visualization not just the model, but I think adding the visualization would help. However, to align the visualization, you'd also need to determine the angle and position of the camera and where the bed is on the image... Seems like a lot. 

This paper, however, seems to showcase that the results could work and could capture even bigger errors like nozzle clogs and more:
https://link.springer.com/article/10.1007/s40964-017-0027-x

bugilt, 2019-06-04:
Most cameras have the specs of the camera somewhere online. You can get the focal length ect... To align the camera to the model. A test print in high contrast to the bed could be used for solving the camera position.

CptanPanic, 2019-08-09:
Well you could print a calibration print like a cube or something that the software would know the size so it could calibrate itself.

kennethjiang, 2019-08-10:
> Well you could print a calibration print like a cube or something that the software would know the size so it could calibrate itself.

Definitely sounds doable.

kennethjiang, 2019-12-19:
I'm closing this issue since this approach, although doable in theory, really seems like a long shot from what we have right now. Maybe a separate project?

closed 2019-12-19 by kennethjiang