TheSpaghettiDetective/obico-server #71: safe upgrading

mcmasterp, 2019-05-30:
How would you advise one to update the server without loosing settings and things? is there anything that should be backed up and restored?

kennethjiang, 2019-05-30:
Not sure if you have tried the instructions described in https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#operating-and-maintaining-the-spaghetti-detective-server. If they don't work for you definitely let me know.

It's probably prudent, prior to upgrade and on regular basis, to back up data base files. But I don't think it's a necessary step of server upgrade.

mcmasterp, 2019-05-30:
oh I did see that but I assumed that it would overwrite files I had altered during install like the config, etc.

mcmasterp, 2019-05-30:
I tried to pull and i got this:

paul@TSD:~/TheSpaghettiDetective$ git pull
error: Pulling is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.

I feel like its related to the docker-compose.override.yml file you created for me to get it working on the nvidia gpu. I'm not so experienced with linux, docker, or git so I don't want to make a mistake. Should I run:

git add docker-compose.override.yml

?

wwsean08, 2019-05-30:
probably not, try running git status and posting the output, I have a decent experience with git/docker so can probably help out a bit.

mcmasterp, 2019-05-31:
@wwsean08 thanks so much. here's the output:

On branch master
Your branch is up to date with 'origin/master'.

Unmerged paths:
  (use "git reset HEAD <file>..." to unstage)
  (use "git add <file>..." to mark resolution)

        both modified:   docker-compose.yml

no changes added to commit (use "git add" and/or "git commit -a")

kennethjiang, 2019-05-31:
@mcmasterp  You can run `git diff` to find out the changes you made in `docker-compose.yml` and move them to `docker-compose. override .yml`. *After that* (don't do it until you have all the changes transferred or else you will lose them), you can do `git checkout docker-compose.yml` and `git pull` again which should work this time.

mcmasterp, 2019-05-31:
thanks @kennethjiang, ok I moved my changes too the override file ( including my email settings, I figured that can't hurt) and ran the git checkout but i get this error:
   error: path 'docker-compose.yml' is unmerged

kennethjiang, 2019-05-31:
Sorry - you were in the middle of a merge. I think this command will work: `git merge --abort && git checkout docker-compose.yml && git pull`

mcmasterp, 2019-06-01:
I tried that command string as well as it minus the merge abort part, you can see the errors:

paul@TSD:~/TheSpaghettiDetective$ git merge --abort && git checkout docker-compose.yml && git pull
fatal: There is no merge to abort (MERGE_HEAD missing).
paul@TSD:~/TheSpaghettiDetective$ git checkout docker-compose.yml && git pull
error: path 'docker-compose.yml' is unmerged

wwsean08, 2019-06-01:
You could try `git checkout -- docker-compose.yml` which should tell git to checkout the original version of it

mcmasterp, 2019-06-01:
same result. Sorry if this because I'm an idiot lol

```
paul@TSD:~/TheSpaghettiDetective$ git checkout -- docker-compose.yml
error: path 'docker-compose.yml' is unmerged
paul@TSD:~/TheSpaghettiDetective$
```

wwsean08, 2019-06-01:
Nope I don't think it's because you're an idiot, this is an interesting problem, I'm out right now, but one thought, can you run `cat docker-compose.yml`, get rid of anything sensitive like the email/password info and post it, I have a theory there are some things that I'd expect to see in a merge in the file.

mcmasterp, 2019-06-01:
`paul@TSD:~/TheSpaghettiDetective$ cat docker-compose.yml
version: '2.4'

x-web-defaults: &web-defaults
  restart: always
  build:
    context: web
  volumes:
    - ./web:/app
  depends_on:
    - redis
  environment:
    EMAIL_HOST:      # -> such as smtp.gmail.com
    EMAIL_HOST_USER:    # -> such as your email address for a Gmail account
    EMAIL_HOST_PASSWORD:     # -> your email account password
    EMAIL_PORT:  
    EMAIL_USE_TLS: 'False'
    DEFAULT_FROM_EMAIL:  
    DEBUG: 'False'    # Don't set DEBUG to True, otherwise the static files will be cached in browser until hard-refresh
    SITE_USES_HTTPS: 'False' # set it to 'True' if https is set up
    SOCIAL_LOGIN: 'False'
    REDIS_URL: 'redis://redis:6379'
    DATABASE_URL: 'sqlite:////app/db.sqlite3'
    INTERNAL_MEDIA_HOST: 'http://web:3334'
    ML_API_HOST: 'http://ml_api:3333'

    #### Optional env vars below this line ###
    # TWILIO_ACCOUNT_SECURITY_API_KEY:
    # TWILIO_ACCOUNT_SID:
    # TWILIO_AUTH_TOKEN:
    # TWILIO_FROM_NUMBER:
    # ENALBE_HISTORY:
    # ML_API_TOKEN:
    # SENTRY_DSN:

services:
  ml_api:
    hostname: ml_api
    restart: always
    build:
      context: ml_api
    environment:
        DEBUG: 'True'
        FLASK_APP: 'server.py'
        # ML_API_TOKEN:
        HAS_GPU: 'True'
    command: bash -c "gunicorn --bind 0.0.0.0:3333 --workers 2 wsgi"

  web:
    <<: *web-defaults
    hostname: web
    ports:
      - "3334:3334"
    depends_on:
      - ml_api
    command: bash -c "python manage.py collectstatic --noinput -c && python manage.py migrate && python manage.py runserver 0.0.0.0:3334"

  tasks:
    <<: *web-defaults
    hostname: tasks
    restart: always
    command: bash -c "celery -A config worker -l info"

  redis:
    restart: always
    image: redis:5.0
`

wwsean08, 2019-06-01:
Sadly nothing out of the ordinary there, you could try a git reset on the file like it mentioned in the output but I'm not sure what's causing the issue so I'm not sure that's not going to cause more problems

kennethjiang, 2019-06-03:
sorry I was out for the weekend. Can you run `git status` and post the result? @mcmasterp

mcmasterp, 2019-06-03:
@wwsean08 thanks for trying, I appreciate it a lot.
@kennethjiang here's the output:

```

paul@TSD:~/TheSpaghettiDetective$ git status
On branch master
Your branch is up to date with 'origin/master'.

Unmerged paths:
  (use "git reset HEAD <file>..." to unstage)
  (use "git add <file>..." to mark resolution)

        both modified:   docker-compose.yml

no changes added to commit (use "git add" and/or "git commit -a")
paul@TSD:~/TheSpaghettiDetective$

```

kennethjiang, 2019-06-03:
Here is what I'd do:

1. `git diff` to see what you have changed in `docker-compose.yml`. Take note of them.
2. `git reset --hard` to reset your git repo.
3. `git pull` to upgrade to the latest code
4. Transfer the changes noted from step 1 to `docker-compose.override.yml`.

let me know if these steps will work.

mcmasterp, 2019-06-03:
@kennethjiang i get this on reset attempt:

`paul@TSD:~/TheSpaghettiDetective$ git reset --hard                              HEAD is now at 5664de2 Set minimum time-lapse duration to be 5 min`

kennethjiang, 2019-06-04:
That looked correct. Did `git pull` run successfully?

mcmasterp, 2019-06-04:
@kennethjiang oh I thought that was an error and didn't go any farther. my mistake, I just did a pull and yes it worked. thank you so much, I'm really excited to have images as email attachments so they will work for us self hosters

closed 2019-06-04 by kennethjiang