TheSpaghettiDetective/obico-server #75: freezes on building tasks

mcmasterp, 2019-06-04:
I recently got past an issue updating tsd (issue #71 I believe) and after a git pull i ran the 

`sudo docker-compose up --build -d`

command and it went well until it got to building tasks and it stopped moving. I waited over 12hrs just to be 100% sure. I had to restart my ssh session to get back to a command prompt and I tried rerunning the build but it immediately starts at the building tasks step and gets stuck there. I tried starting the server with 

`sudo docker-compose up`

but it throws and torrent of errors which is no surprise. What do you recommend I do as a next step?

mcmasterp, 2019-06-04:
update: I removed my compose.override file from the directory and reran the build command and now it seems to get stuck on building web

mcmasterp, 2019-06-04:
update 2: after about 30min it came back with this error:
```

Traceback (most recent call last):
  File "site-packages/docker/utils/build.py", line 97, in create_archive
  File "tarfile.py", line 1980, in addfile
  File "tarfile.py", line 252, in copyfileobj
  File "tempfile.py", line 485, in func_wrapper
OSError: [Errno 28] No space left on device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "bin/docker-compose", line 6, in <module>
  File "compose/cli/main.py", line 71, in main
  File "compose/cli/main.py", line 127, in perform_command
  File "compose/cli/main.py", line 1080, in up
  File "compose/cli/main.py", line 1076, in up
  File "compose/project.py", line 475, in up
  File "compose/service.py", line 342, in ensure_image_exists
  File "compose/service.py", line 1082, in build
  File "site-packages/docker/api/build.py", line 154, in build
  File "site-packages/docker/utils/build.py", line 31, in tar
  File "site-packages/docker/utils/build.py", line 100, in create_archive
OSError: Can not read file in context: /home/paul/TheSpaghettiDetective/web/static_build/media/tsd-pics/raw/8/1559354638.jpg
[10538] Failed to execute script docker-compose

```

mcmasterp, 2019-06-04:
It seems like the web building fails after it exhausts my drive space with temp files

mcmasterp, 2019-06-04:
after a reboot most of the temp files were cleared away so I reran the build and it got past the building web and is currently stuck on building tasks.

kennethjiang, 2019-06-04:
I think I know what the problem was: `web/static_build/media` was too big. If you don't mind losing image files and timelapse files, you can just do `rm -rf web/static_build/media/*` and then retry.

This is related to #2

mcmasterp, 2019-06-04:
ok it finally finished and all seemed fine if not strange but whatever right. Now I'm having another issue after this one but i'll open a new ticket for that

kennethjiang, 2019-08-02:
Closing this issue as it is related to #2

closed 2019-08-02 by kennethjiang