TheSpaghettiDetective/obico-server #80: Notification enhancement

vr6syncro, 2019-06-30:
Hello,

maybe it is possible to add more notification options.
my favourite is telegram. a simple bot  and done.

no more sms cost and better integration in many smart home systems.

ajdavids, 2019-07-05:
Besides a telegram bot, what other notification options would be useful but not overwhelming?

kennethjiang, 2019-07-05:
pushbullet was proposed in #52

vr6syncro, 2019-07-06:
maybe this help u: https://github.com/python-telegram-bot/python-telegram-bot

ajdavids, 2019-07-06:
@vr6syncro I'd already started looking at the telegram bot APIs 😃. Also going to look into the pushbullet and slack bot APIs. Waiting for permission from work to contribute before I start looking at the issue in more depth, though, Thanks for the link.

kennethjiang, 2019-08-02:
Pushbullet is now added by #83 . Kudos to @ajdavids for his awesome PR.

@vr6syncro  Feel free to open another issue for telegram support so that we can attack them one at a time.

closed 2019-08-02 by kennethjiang

flippedcracker, 2019-08-19:
I know this is closed, but I'd like to see Pushover available for notifications. 

https://pushover.net/