TheSpaghettiDetective/obico-server #82: Temperature not being reported correctly for all four metrics

umamiking, 2019-07-31:
**Steps to Reproduce**
1. Install TSD plugin in Octoprint, register your machine and start a print
2. Confirm that the printer has started printing and that the temperature of the extruder and bed is stable. Note these values. In my case it is 205C hotend and 60C on the bed.
3. Login to TSD and look at the in progress job

**Observed**
In my case, it's showing 25.1C bed and 25.6C hot end with both Targets at 0C

**Desired**
The temperatures should be reporting accurate values. TSD should be able to draw these values since OctoprintAnywhere works accurately and Octoprint itself is using correct values. Also, the print is progressing, which would not be possible with a 25.6C extruder temperature.

kennethjiang, 2019-07-31:
Are the temperatures still wrong when you are not printing? Can you try to manually set the bed temperature and see if it is correctly displayed on TSD @umamiking

umamiking, 2019-07-31:
Values are not incorrect when idle. I also set new targets and TSD shows the target and actual values more or less correctly.

kennethjiang, 2019-08-01:
The fix is included in 0.7.1. Please upgrade it when prompted by OctoPrint and let me know if the problem is still not solved.

closed 2019-08-01 by kennethjiang