TheSpaghettiDetective/obico-server #84: No image on mu SpaghettiDetective server

daredevil-ferreira, 2019-08-02:
On my http://<server>:3334/printers/ URL I cannot see any image (I have 2 printers working).

The HTML points to this URL : ="http://example.com/media/tsd-pics/tagged/1/1564726009.jpg"

What is the example.com URL ? If I use http://<server>:3334/media/tsd-pics/tagged/1/1564726009.jpg, I can see the snapshot.

Regards.
David

kennethjiang, 2019-08-02:
You missed a step at configuring the server: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/README.md#configure-django-site

Closing it for now and feel free to re-open it if you still have problem after configuring the site in django admin.

closed 2019-08-02 by kennethjiang

daredevil-ferreira, 2019-08-02:
Yes you are right. If works now
 Thanks

Regards
Davod