TheSpaghettiDetective/obico-server #89: web server keeps restarting

keawepoo, 2019-08-19:
I just installed everything. On trying to access the web page it kicks back saying that there isnt anything located there... After a big of digging I think it might have something to do with this? (see image)
[https://i.imgur.com/p5Sa0Pm.png](https://i.imgur.com/p5Sa0Pm.png)

Thank you.

kennethjiang, 2019-08-19:
It looks like both `web` and `tasks` containers keep on restarting. can you do `docker-compose log -f web` and `docker-compose log -f tasks` to check if there are any errors?

keawepoo, 2019-08-19:
No such command: log
[https://i.imgur.com/8dZxHdO.png](https://i.imgur.com/8dZxHdO.png)

kennethjiang, 2019-08-20:
Sorry, it is `docker-compose logs -f tasks`

keawepoo, 2019-08-20:
A ton of errors... Do I need to install python?
[https://i.imgur.com/JmwmRAv.jpg](https://i.imgur.com/JmwmRAv.jpg)

kennethjiang, 2019-08-20:
Can you try `docker-compose down && docker-compose up -d` and see what happens?

kennethjiang, 2019-08-22:
@keawepoo Do you still need another help on this issue?

keawepoo, 2019-08-23:
@kennethjiang Sorry I didnt see the emails....
[https://i.imgur.com/yIKiQZr.jpg](https://i.imgur.com/yIKiQZr.jpg)

jsiemek, 2019-08-23:
I've same problem.
Ubuntu 18.04.3
docker-ce  version 19.03.1, build 74b1e89

thespaghettidetective_web_1 log:

```
django.db.utils.OperationalError: no such table: django_site,
,
0 static files copied to '/app/static_build', 202 unmodified, 446 post-processed.,
Traceback (most recent call last):,
  File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 85, in _execute,
    return self.cursor.execute(sql, params),
  File "/usr/local/lib/python3.6/site-packages/django/db/backends/sqlite3/base.py", line 298, in execute,
    return Database.Cursor.execute(self, query, params),
sqlite3.OperationalError: no such table: django_site,
,
The above exception was the direct cause of the following exception:,
,
Traceback (most recent call last):,
  File "manage.py", line 15, in <module>,
    execute_from_command_line(sys.argv),
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line,
    utility.execute(),
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 375, in execute,
    self.fetch_command(subcommand).run_from_argv(self.argv),
  File "/usr/local/lib/python3.6/site-packages/django/core/management/base.py", line 316, in run_from_argv,
    self.execute(*args, **cmd_options),
  File "/usr/local/lib/python3.6/site-packages/django/core/management/base.py", line 350, in execute,
    self.check(),
  File "/usr/local/lib/python3.6/site-packages/django/core/management/base.py", line 379, in check,
    include_deployment_checks=include_deployment_checks,,
  File "/usr/local/lib/python3.6/site-packages/django/core/management/commands/migrate.py", line 60, in _run_checks,
    issues.extend(super()._run_checks(**kwargs)),
  File "/usr/local/lib/python3.6/site-packages/django/core/management/base.py", line 366, in _run_checks,
    return checks.run_checks(**kwargs),
  File "/usr/local/lib/python3.6/site-packages/django/core/checks/registry.py", line 71, in run_checks,
    new_errors = check(app_configs=app_configs),
  File "/usr/local/lib/python3.6/site-packages/django/core/checks/urls.py", line 13, in check_url_config,
    return check_resolver(resolver),
  File "/usr/local/lib/python3.6/site-packages/django/core/checks/urls.py", line 23, in check_resolver,
    return check_method(),
  File "/usr/local/lib/python3.6/site-packages/django/urls/resolvers.py", line 396, in check,
    for pattern in self.url_patterns:,
  File "/usr/local/lib/python3.6/site-packages/django/utils/functional.py", line 37, in __get__,
    res = instance.__dict__[self.name] = self.func(instance),
  File "/usr/local/lib/python3.6/site-packages/django/urls/resolvers.py", line 533, in url_patterns,
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module),
  File "/usr/local/lib/python3.6/site-packages/django/utils/functional.py", line 37, in __get__,
    res = instance.__dict__[self.name] = self.func(instance),
  File "/usr/local/lib/python3.6/site-packages/django/urls/resolvers.py", line 526, in urlconf_module,
    return import_module(self.urlconf_name),
  File "/usr/local/lib/python3.6/importlib/__init__.py", line 126, in import_module,
    return _bootstrap._gcd_import(name[level:], package, level),
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked,
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module,
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed,
  File "/app/config/urls.py", line 21, in <module>,
    path('', include('app.urls')),,
  File "/usr/local/lib/python3.6/site-packages/django/urls/conf.py", line 34, in include,
    urlconf_module = import_module(urlconf_module),
  File "/usr/local/lib/python3.6/importlib/__init__.py", line 126, in import_module,
    return _bootstrap._gcd_import(name[level:], package, level),
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked,
[... 117 lines cut ...]
    return getattr(self.get_queryset(), name)(*args, **kwargs),
  File "/usr/local/lib/python3.6/site-packages/django/db/models/query.py", line 605, in first,
    for obj in (self if self.ordered else self.order_by('pk'))[:1]:,
  File "/usr/local/lib/python3.6/site-packages/django/db/models/query.py", line 268, in __iter__,
    self._fetch_all(),
  File "/usr/local/lib/python3.6/site-packages/django/db/models/query.py", line 1186, in _fetch_all,
    self._result_cache = list(self._iterable_class(self)),
  File "/usr/local/lib/python3.6/site-packages/django/db/models/query.py", line 54, in __iter__,
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size),
  File "/usr/local/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1065, in execute_sql,
    cursor.execute(sql, params),
  File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 68, in execute,
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute),
[... cut ...]
  File "/usr/local/lib/python3.6/site-packages/django/db/utils.py", line 89, in __exit__,
    raise dj_exc_value.with_traceback(traceback) from exc_value,
  File "/usr/local/lib/python3.6/site-packages/django/db/backends/utils.py", line 85, in _execute,
    return self.cursor.execute(sql, params),
  File "/usr/local/lib/python3.6/site-packages/django/db/backends/sqlite3/base.py", line 298, in execute,
    return Database.Cursor.execute(self, query, params),
django.db.utils.OperationalError: no such table: django_site,
```

edsai, 2019-08-23:
I've got the same issue.

kennethjiang, 2019-08-23:
Thank you folks for reporting this bug and providing the info. It is a bug we introduced recently. Please pull the latest and it should work now.

commit 246826fd (2019-08-23): Closes #89. Can not access site in module code as the migration has not run if this is the first time server starts
closed 2019-08-23 by kennethjiang