TheSpaghettiDetective/obico-server #99: Allow for custom camera URLs

trunzoc, 2019-09-30:
I have a weird setup where I run a webcam through an IPCam DVR package (Blue Iris).  BI then feeds the webcam, with added timestamp and a watermark, to wherever I need it, including my OctoPrint instance.

The issue is that the timestamp keeps my prints in the yellow most of the time.

I would would be super helpfull if I could plug a URL to directly pull from the webcam itself, instead of using the process image that OctoPrint uses.

trunzoc, 2019-09-30:
[image] 
postprocess with watermark and timestamp

[image]
Original webcam feed

trunzoc, 2019-09-30:
Original Webcam feed...
[image]

Postprocessed with Watermark and timestamp...
[image]

RobeeeJay, 2020-09-24:
Right now I have an issue with my Pi 4 re-encoding the video stream from the Pi 2.1 webcam, which eats a lot of CPU. I'm running TSD on a Jetson-Nano which has two camera ports, it would be wonderful if TSD could get the video feed from somewhere other than the Octoprint plugin, then I wouldn't have to use the Pi to re-encode everything and send it over the network.

I think this is the same request you are making, so I thought I would add it here!

raymondh2, 2020-11-11:
Custom URLs have been supported for a time now. Closing issue

closed 2020-11-11 by raymondh2

RobeeeJay, 2020-11-14:
I might be being stupid, but I can't see anywhere I can specify a webcam URL in either TSD plugin in OctoPrint or on my Jetson Nano TSD server. 😞

RobeeeJay, 2020-11-14:
I will answer my own question here, the custom URLs come from OctoPrint so you can only change it there.

timur-tabi, 2022-04-19:
I would like to reopen this issue.  I think it's a big limitation that TSD can only use the default camera configured in Octoprint.  Instead, I really need to be able to have TSD use an alternative camera that I specify via a streaming URL.  In my case, I would like to use a dedicated IP camera that is not known to Octoprint.

I also have a Jetson Nano configured for TSD.

kennethjiang, 2022-04-20:
@timur-tabi what you mean is probably this issue: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/98. I know it's been open for a long time. But there are some technical difficulties that have prevented us from implementing it. Maybe with Pi 4b being common now it can be revisited.

ttabi, 2022-04-20:
I don't need multiple cameras, just one.  If anything, I would say that this bug is a per-requisite for #98 -- you can't support multiple cameras without supporting an IP camera first.

However, I don't see how this has anything to do with a Raspberry Pi.  TSD is running on my Nano.  There's no reason why TSD couldn't directly stream the video to itself, bypassing Octoprint entirely.  This is how video streams work today anyway.  Octoprint just provides the stream URL to the browser, and the browser does the actual streaming.

kennethjiang, 2022-04-20:
@timur-tabi any reason why you can't set the snapshot url in octoprint? TSD plugin's basic streaming takes from the snapshot url in octoprint.

ttabi, 2022-04-20:
Sure, I can do that, but the real problem is that TSD is only capable of using the *primary* camera.  I suppose I can just specify a completely different URL.

If TSD is capable of using any snapshot URL, then why not just let me specify that URL in the TSD plugin configuration page?  

Ideally, I would like to be able to plug in a USB camera into the Nano itself, eliminating Octoprint as the middleman.

kennethjiang, 2022-04-21:
The reason why I'm hesitating to add a URL is it should be part of the multi-camera solution. 

The failing detection relies on many APIs that OctoPrint supplies to be able to function (the thousands of lines of code in the plugin are not useless lol). It's possible that you fork the code and change it to make it work for your specific setup. But I don't see how this can be applicable to other users. But if you do figure this out, feel free to send a PR and I'll review it.

timur-tabi, 2022-04-21:
Does the plugin do processing on the image before passing it to the server?

It seems that adding a URL is much, much simpler than adding multi-camera support.  I don't care about multi-camera support, but I would love to be able to specify my a different URL.