TheSpaghettiDetective/obico-server #100: Private server - mails not being sent

Wireheadbe, 2019-10-06:
Hi, I'm running a private docker instance and I'm testing the notification functions.
It seems i'm not getting anything out. Google SMTP with TLS is defined, however; upon running:

```
In [1]: from app.notifications import send_failure_alert_email
In [2]: send_failure_alert_email(Printer.objects.first(), False, False)
```

No error is being thrown. 

On the host running the docker containers - I'm running TCPdump, not a single packet is detected on port 587:
```
sudo tcpdump -i eth0 -n tcp port 587
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
```

`docker-compose logs -f web` doesn't show anything as well that very moment.

Any idea where I can look?

kennethjiang, 2019-10-07:
This is possibly a bug. If you want you can help us debug it. You can set a breakpoint in `send_failure_alert_email` and step through to see if this statement is called: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/web/app/notifications.py#L68

Wireheadbe, 2019-10-07:
Actually I already did that. I printed the contents of msg via print() and could see some nice html. I also tried defining a telegram bot, and the same behaviour ensues. The telegram message "supposedly" is sent correctly, but nothing gets out. Could it be docker itself? Routing?

kennethjiang, 2019-10-07:
What if you configure a wrong hostname/port for the smtp server or telegram server? I remember seeing error messages for telegram bot when telegram server is not reachable. But I don't remember if smtp will just silently fail.

Wireheadbe, 2019-10-08:
OK, I changed the docker-compose.yml and set the SMTP hostname to my iptables router/fw, which isn't running any mail daemon:

- `EMAIL_HOST: '192.168.100.1'`

And then ran:

- `sudo docker-compose up --build -d`

To rebuild the docker containers. And then connect to the web container via

- `sudo docker-compose exec web bash`

And execute
- `python manage.py shell_plus`

And then try to send a mail:
```
In [1]: from app.notifications import send_failure_alert_email
In [2]: send_failure_alert_email(Printer.objects.first(), False, False)
In [3]:

```
I get no error, nor does it seem slow in trying to access the server, altough the server is invalid

Wireheadbe, 2019-10-08:
Ok; I know why it's behaving that way. Even in a private instance, it needs my e-mail verified, otherwise the 'emails' variable is empty, causing it to bail out.

Also; telegram integration has a similar issue. CSP is being used on the Telegram side; which causes it to be un-embeddable on an instance running on port 3334. It must be run on e.g. port 80 or 443.

kennethjiang, 2019-10-08:
Thank you for debugging this for us! Will be great if you can send a PR.

Wireheadbe, 2019-10-08:
I think the variable ACCOUNT_EMAIL_VERIFICATION in notifications.py takes care of that, no? Maybe best to pre-fill this in docker-compose.yml with some comment about what it does?

Also, maybe add the comment in that very same file that by enabling telegram bots; you implicitly need a reverse proxy to either run it on standard http or https; otherwise oauth will fail. I'm currently rebuilding from scratch to check some other possible issues related to these kind of edge cases

kennethjiang, 2019-12-19:
@Wireheadbe I know multiple people have been able to get email work on their private servers. But what you said also sounds reasonable. If you want to send in a PR for what you changed, I'll be happy to review and merge it.

jreed1701, 2020-01-10:
I found this issue after a desire to test my email settings as well and what's needed is two things:

1. The docker-compose.yml is not the place to configure email servers and settings. This should be in the user profile or admin profile.
2. A GUI interface to test the email settings tool. Right now all you can do is follow the procedure outlined here or wait for the system to detect a failure and hope it works.

Does anyone else agree? I can make issues against this.  Maybe even work it myself I my ample spare time haha...

RaymondHimle, 2020-02-03:
@kennethjiang Close due to inactivity

jreed1701, 2020-02-03:
I think the real issue is ability to check if your email settings are correct or not.

kennethjiang, 2020-02-04:
@jreed1701 I don't think email settings should be in user profile. It's a common practice to use consistent email settings across the whole site. But email testing tool is a good idea.

@jreed1701 feel free to open another issue for the email testing tool. Closing this one for now

closed 2020-02-04 by kennethjiang

jreed1701, 2020-02-04:
@kennethjiang Sounds good. I might implement if i can find the time.

kennethjiang, 2020-02-05:
Thank you @jreed1701