TheSpaghettiDetective/obico-server #103: Local server - adding printer fails

mplogas, 2019-10-25:
**System:**

TheSpaghettiDetective: latest from master
Docker: 19.03.1
Docker-Compose: 1.24.1
OS: Debian 9
Kernel: 4.15.18 x64

**Description:**

Adding a printer seems pretty straight forward, but the "current print" field is required - which is only an empty field for me. Thus adding my printer (and an auth token) isn't possible.

**Expected behaviour:**

Since there can't be any "current print" when adding the first printer to the system, I was expecting this field NOT to be mandatory and therefore allwoing me to set up the first printer.

**Steps to reproduce**

1. Set up TheSpaghettiDetective on-premise, according to the documentation (using docker etc)
2. Set up a user and a site.
3. Set up the first printer.
	a. put in name
	b. select user from dropdown
	c. create an auth token
	d. hit save
	e. an error message appears

[image]

kennethjiang, 2019-10-25:
It's strange. [`current_print` is not a required field in the model](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/web/app/models.py#L128).

In any case, you can (actually are supposed to) create a printer from the normal user interface at http://yourserver:3334/printers/. Is there a reason why it is not an option for you?

mplogas, 2019-10-26:
@kennethjiang regarding `current_print` it's still that issue when going through the django admin site. I did not knew about the regular interface, this works very well. Thanks!

kennethjiang, 2019-10-26:
Great. Thank you for confirming that!

closed 2019-10-26 by kennethjiang