TheSpaghettiDetective/obico-server #115: New installation getting error 500 on web1

luismiparra, 2019-11-11:
Hi,

I installed a new instance of the server on a debian10 docker+dockercompose

When I try to access the site I get a err 500

This is the log of docker web1

`
django.channels.server ERROR    HTTP GET /accounts/login/?next=/printers/ 500 [0.03, 10.5.1.2:57970],
django.channels.server INFO     HTTP GET / 302 [0.00, 10.5.1.2:57970],
django.channels.server INFO     HTTP GET /printers/ 302 [0.00, 10.5.1.2:57970],
django.request ERROR    Internal Server Error: /accounts/login/,
Traceback (most recent call last):,
  File "/usr/local/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner,
    response = get_response(request),
  File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 156, in _get_response,
    response = self.process_exception_by_middleware(e, request),
  File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 154, in _get_response,
    response = response.render(),
  File "/usr/local/lib/python3.6/site-packages/django/template/response.py", line 106, in render,
    self.content = self.rendered_content,
  File "/usr/local/lib/python3.6/site-packages/django/template/response.py", line 83, in rendered_content,
    content = template.render(context, self._request),
  File "/usr/local/lib/python3.6/site-packages/django/template/backends/django.py", line 61, in render,
    return self.template.render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 169, in render,
    with context.bind_template(self):,
  File "/usr/local/lib/python3.6/contextlib.py", line 81, in __enter__,
    return next(self.gen),
  File "/usr/local/lib/python3.6/site-packages/django/template/context.py", line 242, in bind_template,
    processors = (template.engine.template_context_processors +,
  File "/usr/local/lib/python3.6/site-packages/django/utils/functional.py", line 37, in __get__,
    res = instance.__dict__[self.name] = self.func(instance),
  File "/usr/local/lib/python3.6/site-packages/django/template/engine.py", line 85, in template_context_processors,
    return tuple(import_string(path) for path in context_processors),
  File "/usr/local/lib/python3.6/site-packages/django/template/engine.py", line 85, in <genexpr>,
    return tuple(import_string(path) for path in context_processors),
  File "/usr/local/lib/python3.6/site-packages/django/utils/module_loading.py", line 17, in import_string,
    module = import_module(module_path),
  File "/usr/local/lib/python3.6/importlib/__init__.py", line 126, in import_module,
    return _bootstrap._gcd_import(name[level:], package, level),
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import,
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load,
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked,
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked,
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module,
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed,
  File "/app/app/context_processors.py", line 2, in <module>,
    from app_ent.detective_hour_accounting import dh_is_unlimited,
ModuleNotFoundError: No module named 'app_ent',
django.channels.server ERROR    HTTP GET /accounts/login/?next=/printers/ 500 [0.03, 10.5.1.2:57970],
`

kennethjiang, 2019-11-12:
it's a known bug. will be fixed in the next couple days. Stay tuned.

kennethjiang, 2019-11-12:
@luismiparra Can you pull it from master and try again?

closed 2019-11-12 by kennethjiang

luismiparra, 2019-11-12:
It works :) !! 

Thanks a lot!

I love your project

commit 527065e7 (2019-11-12): Closes #115