TheSpaghettiDetective/obico-server #119: Download Time Lapses

yannickh2, 2019-11-22:
Hi all,

I love this project. I saw the question about downloading time lapses in Discord and did not find an open issue so I thought I open one.

Use - Case: We are a start-up and prototyping a vineyard sensor solution and just printing the case. I want to write a blog article and make tweets about it. For that a GIF of the 3D print process would be a nice visual. It would be great to download a video and I can convert the video into GIF myself or something.

Thank you.

Best,
Yannick

RaymondHimle, 2019-12-26:
This is now an option in the latest version of TSD.

kennethjiang, 2019-12-26:
You are right @RaymondHimle . :)

I'm closing this issue as it is now available in the latest TSD version @yannickh2

closed 2019-12-26 by kennethjiang