TheSpaghettiDetective/obico-server #123: Docker build failing on Pillow

marcelrv, 2019-12-01:
Docker build fails on building Pillow

`docker-compose up --build -d`
Seems zlib issue in pillow.

building on Linux docker-server 5.0.0-36-generic #39~18.04.1-Ubuntu SMP Tue Nov 12 11:09:50 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```
Requirement already satisfied: vine>=1.1.3 in /usr/local/lib/python3.6/site-packages (from amqp<3.0,>=2.4.0->kombu<5.0,>=4.2.0->celery[redis]==4.2.1->-r requirements.txt (line 25)) (1.3.0)
Building wheels for collected packages: Pillow, django-jstemplate, pytelegrambotapi
  Building wheel for Pillow (setup.py): started
  Building wheel for Pillow (setup.py): finished with status 'error'
  ERROR: Command errored out with exit status 1:
   command: /usr/local/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-n6m0fz2t/Pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-n6m0fz2t/Pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-8ayjm2vg --python-tag cp36
       cwd: /tmp/pip-install-n6m0fz2t/Pillow/
  Complete output (172 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.6
  creating build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageStat.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/_tkinter_finder.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/GimpGradientFile.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/BufrStubImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/PcdImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/__main__.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageMorph.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/GdImageFile.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/MpoImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/JpegPresets.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageWin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImagePath.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageFile.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageColor.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/TiffImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/GribStubImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/GifImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/CurImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/XpmImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/FtexImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/MspImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/PixarImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/PdfImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageFont.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageDraw2.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageEnhance.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/DdsImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/WalImageFile.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageShow.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/IptcImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageGrab.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageMode.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageTk.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/PcxImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/__init__.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/TiffTags.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/BdfFontFile.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/IcoImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/XbmImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/DcxImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/PyAccess.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageCms.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImageFilter.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/_binary.py -> build/lib.linux-x86_64-3.6/PIL
  copying src/PIL/ImtImagePlugin.py -> build/lib.linux-x86_64-3.6/PIL
[... 305 lines cut ...]

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-n6m0fz2t/Pillow/setup.py", line 902, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:

    The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.

    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html

    ----------------------------------------
  Rolling back uninstall of Pillow
  Moving to /usr/local/lib/python3.6/site-packages/PIL/
   from /usr/local/lib/python3.6/site-packages/~IL
  Moving to /usr/local/lib/python3.6/site-packages/Pillow-5.4.1.dist-info/
   from /usr/local/lib/python3.6/site-packages/~illow-5.4.1.dist-info
ERROR: Command errored out with exit status 1: /usr/local/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-n6m0fz2t/Pillow/setup.py'"'"'; __file__='"'"'/tmp/pip-install-n6m0fz2t/Pillow/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-fhkxiy4d/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
ERROR: Service 'web' failed to build: The command '/bin/sh -c pip install -r requirements.txt' returned a non-zero code: 1

```

kennethjiang, 2019-12-01:
@marcelrv We can't reproduce this ourselves. 

Can you follow the steps we listed in discord channel to do more debugging?

marcelrv, 2019-12-01:
issue seems to be with cb93323d910506bce88146f275284bf73d57b720
Downgrading to older pillow prevents the issue and allows to complete the build

cmsimike, 2019-12-02:
Confirmed the issue locally, and changing to the older version of Pillow seems to fix it.

kennethjiang, 2019-12-02:
Will appreciate it if someone who can reproduce this error find fix it (not by downgrading Pillow as the older version has security vulnerabilities) and submit a PR.

cmsimike, 2019-12-02:
I spent some time looking into this and tried a few different work arounds, but nothing ended up helping. I have to focus on other things for  a while but I wanted to link https://github.com/python-pillow/Pillow/issues/3103 as this seems to be a similar issue we're seeing.

In case anyone wants to play around with the fix, it should be ```DFLAGS=-L/usr/lib/x86_64-linux-gnu/```

If no one picks this up when I have some time, I'll try out this solution and report back.

marcelrv, 2019-12-02:
I saw that one yesterday as well. Did not work
also tried to install zlib-dev  (`RUN apk add zlib-dev`) no success
also tried `RUN LDFLAGS=-L/usr/lib/x86_64-linux-gnu/ pip install --no-binary pillow Pillow==6.2.0`
also tried `RUN LDFLAGS=-L/usr/lib/x86_64-linux-gnu/ pip install --no-binary pillow Pillow==6.2.1`
both no joy....

cmsimike, 2019-12-03:
^ agreed.

kennethjiang, 2019-12-03:
Thank you so much folks for the help!

I'm ok with rolling back pillow version. however it'll be nice if we can stay on a 6.x.x version. If it's not too much work, can one of you try to see if one of the earlier 6.x.x versions would work? Thanks a lot!

cmsimike, 2019-12-03:
So far I've tried 6.1.0 and 6.0.0 but all of them seem to have similar issues. I'll submit a PR to downgrade Pillow to 5.4.1

kennethjiang, 2019-12-03:
Merged! Thanks @cmsimike

closed 2019-12-03 by kennethjiang

marcelrv, 2019-12-04:
thanks @cmsimike @kennethjiang

PR #125 (2019-12-03): Downgrade pillow to avoid build errors
Fixes #123
commit 9162928e (2019-12-03): Downgrade pillow to avoid build errors

Fixes #123
commit 7b4ec372 (2019-12-03): Downgrade pillow to avoid build errors

Fixes #123