TheSpaghettiDetective/obico-server #131: Video from uploaded timelapes wont play.

CamoJackson, 2019-12-11:
After uploading a video to TSD and the file is processed it won't playback. No data is being transferred to the browser.

windows:
Major  Minor  Build  Revision
-----  -----  -----  --------
10     0      17134  0
firewall is off

Docker version 19.03.5, build 633a0ea
TSD from commit :7a6d839347793cd718ad25c09d5dbdb7c8d97ac5

kennethjiang, 2019-12-11:
I tested it and it worked. Can you provide more test data and steps to reproduce?

CamoJackson, 2019-12-11:
I did a fresh install of docker desktop for windows followed by the docker-compose build. then the settings for the site was set up for the guide

Firefox debuge reports 
This page uses the non standard property “zoom”. Consider using calc() in the relevant property values, or using “transform” along with “transform-origin: 0 0”. prints
<source> element has no “src” attribute. Media resource load failed. prints
All candidate resources failed to load. Media load paused. 2 prints
HTTP load failed with status 404. Load of media resource http://example.com/media/tsd-timelapses/private/1.mp4 failed. prints
All candidate resources failed to load. Media load paused. prints
VIDEOJS: ERROR: (CODE:4 MEDIA_ERR_SRC_NOT_SUPPORTED) No compatible source was found for this media. 
Object { code: 4, message: "No compatible source was found for this media." }
video.min.js:12:1480
MouseEvent.mozPressure is deprecated. Use PointerEvent.pressure instead. video.min.js:12:8610
All candidate resources failed to load. Media load paused. 3 prints
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://10.0.25.69/media/tsd-timelapses/private/2_p.json. (Reason: CORS request did not succeed).

Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://10.0.25.69/media/tsd-timelapses/private/1_p.json. (Reason: CORS request did not succeed).

kennethjiang, 2019-12-13:
Sorry I missed your update!

Can you upload the same video to TSD cloud (https://app.thespaghettidetective.com/)  and see if they will play back? Since TSD cloud uses the same code as the private server, we can test if the problem is caused by server environment or those video files by trying to process the same video files in TSD cloud.

CamoJackson, 2019-12-15:
Yes, it does work on the TSD cloud. I have also spun up a centos VM and installed it there, no luck. What version of docker does the TSD cloud run on mabey there was a update to docker that caused the problem

kennethjiang, 2019-12-15:
You can run `docker-compose logs web` and look for errors in the output. The docker version requirements can be found at: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#software-requirements

RaymondHimle, 2020-01-17:
Looks like the Django setup step might have been missed. 
"Load of media resource http://example.com/media/tsd-timelapses/private/1.mp4 failed. "

kennethjiang, 2020-02-08:
Closing because of inactivity.

closed 2020-02-08 by kennethjiang