TheSpaghettiDetective/obico-server #132: TSD causes Wi-Fi to drop out on Pi 4

tusing, 2019-12-13:
As soon as I enabled TSD, my Wi-Fi would frequently drop out on my Pi 4. As soon as I uninstalled and rebooted, my Wi-Fi didn't drop out again.

I know it's not much to go on, but here are errors from dmesg:

```console
[29638.819812] brcmfmac: brcmf_run_escan: error (-110)
[29638.819821] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[29641.369903] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[29643.929960] brcmfmac: _brcmf_set_multicast_list: Setting mcast_list failed, -110
[29646.490004] brcmfmac: brcmf_run_escan: error (-110)
[29646.490014] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[29649.050057] brcmfmac: _brcmf_set_multicast_list: Setting allmulti failed, -110
[29651.610136] brcmfmac: brcmf_run_escan: error (-110)
[29651.610145] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[29654.170180] brcmfmac: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
[29656.730240] brcmfmac: brcmf_run_escan: error (-110)
[29656.730249] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[29664.410384] brcmfmac: brcmf_run_escan: error (-110)
[29664.410393] brcmfmac: brcmf_cfg80211_scan: scan error (-110)
[29840.494189] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[29840.494197] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[91978.174723] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[91978.174730] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
[92054.976281] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[92054.976289] brcmfmac: brcmf_cfg80211_get_channel: chanspec failed (-110)
```

After uninstalling TSD and rebooting my errors ceased.

Let me know if there's any more information I can get to help.

kennethjiang, 2019-12-13:
The correlation between the wifi problem and TSD plugin does sound strange. I can't think of any reason why TSD plugin would cause wifi to drop off. It doesn't mess with network settings at all...

What do you mean by `"OctoPrint Anywhere"-like functionality`? OctoPrint Anywhere doesn't mess with network settings either. And I have never received reports like this from OctoPrint Anywhere users.

RaymondHimle, 2019-12-13:
Only thing I can think of off the top of my head is a bad power supply. Especially if using the streaming function with TSD

tusing, 2019-12-14:
I figured it out.

When the Pi 4 is communicating over Wi-Fi it seems to up the RF noise. The network optimization "features" in my router would then reach their tipping point and disable features like multicast traffic, begin messing with the channels, and whatnot.

It seems TSD + Octoprint would reach the tipping point more often than just Octoprint, but after upping the camera framerate I was able to reach the tipping point with Octoprint alone.

TL;DR issue has nothing to do with TSD itself, but rather with increased Wi-Fi activity messing with my networking gear, which in turn prevented communications from the Pi.

closed 2019-12-14 by tusing

kennethjiang, 2019-12-15:
Wow thank you for sharing this in-depth knowledge with us! I definitely learned!