TheSpaghettiDetective/obico-server #134: Can't access server

ChristophSchranz, 2019-12-17:
Hi,
this is a very nice app.
Unfortunately, I can't access the web service after the docker start.
Steps to reproduce:
```
#clone and cd
#modify docker-compose.yml
git pull && docker-compose up --build -d
curl 127.0.0.1:3334
#>curl: (56) Recv failure: Connection reset by peer
sudo netstat -nlp |grep 3334
#>tcp6       0      0 :::3334                 :::*                    LISTEN      23794/docker-proxy

docker-compose logs | tail -5
web_1     | System check identified no issues (0 silenced).
web_1     | December 17, 2019 - 13:26:09
web_1     | Django version 2.1.11, using settings 'config.settings'
web_1     | Starting ASGI/Channels version 2.1.7 development server at http://0.0.0.0:3334/
web_1     | Quit the server with CONTROL-C.
root@il071:/mnt/glusterfs/dtz/src/TheSpaghettiDetective# docker-compose logs | tail -10
web_1     | daphne.server INFO     HTTP/2 support not enabled (install the http2 and tls Twisted extras)
web_1     | daphne.server INFO     Configuring endpoint tcp:port=3334:interface=0.0.0.0
web_1     | daphne.server INFO     Listening on TCP address 0.0.0.0:3334
web_1     | Performing system checks...
web_1     | 
web_1     | System check identified no issues (0 silenced).
web_1     | December 17, 2019 - 13:26:14
web_1     | Django version 2.1.11, using settings 'config.settings'
web_1     | Starting ASGI/Channels version 2.1.7 development server at http://0.0.0.0:3334/
web_1     | Quit the server with CONTROL-C.
```

my settings:
`version: '2.4'

x-web-defaults: &web-defaults
  restart: unless-stopped
  build:
    context: web
  volumes:
    - ./web:/app
  depends_on:
    - redis
  environment:
    EMAIL_HOST: smtp.gmail.com    # -> such as smtp.gmail.com
    EMAIL_HOST_USER: my@gmail.com   # -> such as your email address for a Gmail account
    EMAIL_HOST_PASSWORD: secret   # -> your email account password
    EMAIL_PORT: 587
    EMAIL_USE_TLS: 'True'
    DEFAULT_FROM_EMAIL: 'changeme@example.com'
    DEBUG: 'False'    # Don't set DEBUG to True, otherwise the static files will be cached in browser until hard-re$
    SITE_USES_HTTPS: 'False' # set it to 'True' if https is set up
    SOCIAL_LOGIN: 'False'
    REDIS_URL: 'redis://redis:6379'
    DATABASE_URL: 'sqlite:////app/db.sqlite3'
    INTERNAL_MEDIA_HOST: 'http://web:3334'
    ML_API_HOST: 'http://ml_api:3333'
`

Can someone please help a fool? 
Is there a setting I've forgot?

ChristophSchranz, 2019-12-17:
It seems that it works on another node. So I'll close this issue as it could be a setup problem on that node.

closed 2019-12-17 by ChristophSchranz

kennethjiang, 2019-12-18:
Glad that you figured this out!

tenumars, 2022-03-19:
wait, so what did you do? having same issue right now