TheSpaghettiDetective/obico-server #135: Not actively watching

ChadOhman, 2019-12-19:
Hi all -

I've followed the directions in the readme to set up the detective however I'm getting "The Detective Is Not Watching (Why?)"

The site address is correct, the plugin says it's connected. I'm getting an image on the 3334 interface.

What am I doing wrong here?

Thanks.

kennethjiang, 2019-12-19:
Are you using TSD cloud (https://www.thespaghettidetective.com/) or your own private server? @ChadOhman

ChadOhman, 2019-12-19:
Oops, sorry, I should have indicated. My own private server. Ubuntu Server 18.04 with the latest docker/docker-compose.

Thanks

RaymondHimle, 2019-12-19:
Give the Octoprint device a full restart (OS level.) This fixes the issue in some cases.

ChadOhman, 2019-12-20:
That worked, thank you!

closed 2019-12-20 by ChadOhman

kennethjiang, 2019-12-20:
Thank you @RaymondHimle for helping out here. Do you know why an OS-level restart will fix it?

RaymondHimle, 2019-12-20:
No I don't know why. But a restart seems to fix a lot of issues here. An OS restart might not have been required for this issue but it doesn't hurt.