TheSpaghettiDetective/obico-server #136: feature-request: url-blueprint for proxy

ChristophSchranz, 2019-12-19:
Hi, 
I really enjoy this app and would like to use it via a proxy. Therefore, I've set up a Apache2 reverse proxy that indirects the webservice, api and websockets of TSD.

Here is the config:
```
		# Proxy for The Spaghetti Detective
                RewriteRule ^/tsd$ printers/ [R]
                RewriteRule ^/tsd/ printers/ [R]
                RewriteRule ^/thespaghettidetective$ printers/ [R]
                RewriteRule ^/thespaghettidetective/ printers/ [R]
                RewriteRule ^/printers$ printers/ [R]
                #SetEnv proxy-nokeepalive 1

                ProxyPass /printers/ http://127.0.0.1:3334/printers/
                ProxyPassReverse /printers/ http://127.0.0.1:3334/printers/
                <Location "/api/v1/printers/">
                    ProxyPass        http://127.0.0.1:3334/api/v1/printers/
                    ProxyPassReverse http://127.0.0.1:3334/api/v1/printers/
                </Location>

                ProxyPass /prints/ http://127.0.0.1:3334/prints/
                ProxyPassReverse /prints/ http://127.0.0.1:3334/prints/
                ProxyPass /media/ http://127.0.0.1:3334/media/
                ProxyPassReverse /media/ http://127.0.0.1:3334/media/
                ProxyPass /user_preferences/ http://127.0.0.1:3334/user_preferences/
                ProxyPassReverse /user_preferences/ http://127.0.0.1:3334/user_preferences/
                ProxyPass /accounts/login http://127.0.0.1:3334/accounts/login
                ProxyPassReverse /acccounts/login http://127.0.0.1:3334/accounts/login

                <Location "/static/">
                    ProxyPass        http://127.0.0.1:3334/static/
                    ProxyPassReverse http://127.0.0.1:3334/static/
                </Location>

                <Location "/ws">
                    ProxyPass        ws://127.0.0.1:3334/ws
                    ProxyPassReverse ws://127.0.0.1:3334/ws
                </Location>
```
Using this, i can access the printer view on `server/printers/`. Unfortunately, this setting needs multiple urls. I would like to use a common prefix, like `server/tsd/printers/`.

How can one achieve this?

Best, 
Chris

kennethjiang, 2019-12-19:
Unfortunately currently the server paths are hard-coded in various `urls.py` files. 

I'm not sure I entirely understand your question though. If you want to add `/server/` in front of the url, you should be able to achieve that in apache config files, right? I'm under the impression this is the common practice when it comes to alerting the server path (as opposed to changing it in Django app).

ChristophSchranz, 2019-12-20:
Thank you @kennethjiang!

kennethjiang, 2019-12-20:
I'm not sure if I helped here... I'm not an expert on Apache config myself (we host TSD using Kubernetes), but let me know if you still have questions.

ChristophSchranz, 2019-12-20:
We are running it now without unique prefixes in the url, so we  have `www.server.com/printers` or `www.server.com/media`. As long as no other service requires the same url (and we can't change it there either) we don't run in a url conflict. :) 
However, maybe, others would also need that url prefix (e.g. if there are legal restrictions to share data to a third-party). It would be great, if that setting would be possible in a longer term!

kylekyle, 2021-04-01:
Removing the hostname from `taggedSrc` in [SteamingBox.vue](/web/frontend/src/common/StreamingBox.vue) would make the image visible from both the internal network and a reverse proxy. So it would be:

`/media/tsd-pics/tagged/4/28/1617242075.675524.jpg` 

instead of 

`http://192.168.0.2:3334/media/tsd-pics/tagged/4/28/1617242075.675524.jpg`

The latter can't be accessed from behind a reverse proxy.