TheSpaghettiDetective/obico-server #148: Printer connected but no webcam stream on TSD web interface

PedroC1999, 2019-12-31:
I have setup TSD as per the instructions in the README.
The printer is communicating with TDS, and I can see temperatures/progress etc on the interface.
The server is not local to my network, but the secret tests are valid.
Within OctoPrint, I can see the webcam feed fine. The camera is setup using MotionEye.
In TDS, I simply see a black screen as show below.

[image]

I am able to provide any logs or debug anything if required,
Many thanks!

closed 2019-12-31 by PedroC1999

kennethjiang, 2019-12-31:
@PedroC1999 Just want to check if you closed it because the problem was fixed? If so, can you let me know what you did to fix it?

PedroC1999, 2019-12-31:
The issue was in the "Site" setup. I had used the Octoprint Servers IP instead of the TSD IP. 

This could possibly be added to the FAQ on the TSD website for the "no webcam stream" section. 

Many thanks!

kennethjiang, 2019-12-31:
Gotcha! It's a really good suggestion to add this to the instructions!